openbmc/openbmc #1565: Set OpenBMC ip address with ipmitool in-band doesn't work

zet809, 2017-05-05:
When trying to set openbmc ip address in-band from the host with ipmitool, the new ip address doesn't take effect at all:
```
[p9euh02 /]# ipmitool lan set 1 ipaddr 10.6.9.200
Setting LAN IP Address to 10.6.9.200
[p9euh02 /]# echo $?
0
[p9euh02 /]# ipmitool lan print 1
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5
                        : User     : MD5
                        : Operator : MD5
                        : Admin    : MD5
                        : OEM      : MD5
IP Address              : 10.6.9.254
Subnet Mask             : 255.0.0.0
MAC Address             : 70:e2:84:14:24:74
Default Gateway IP      : 10.0.0.101
Cipher Suite Priv Max   : Not Available
Bad Password Threshold  : Not Available
```
The new ip address doesn't work at all:
```
[root@stratton01 bin]# ping 10.6.9.200 -c 1
PING 10.6.9.200 (10.6.9.200) 56(84) bytes of data.
From 10.6.29.1 icmp_seq=1 Destination Host Unreachable

--- 10.6.9.200 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms
[root@stratton01 bin]# ping 10.6.9.254 -c 1
PING 10.6.9.254 (10.6.9.254) 56(84) bytes of data.
64 bytes from 10.6.9.254: icmp_seq=1 ttl=64 time=0.366 ms

--- 10.6.9.254 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 0.366/0.366/0.366/0.000 ms
```

williamspatrick, 2017-05-05:
I think we require an extra step to "finalize" any ip changes. This is to avoid someone changing the IP address or gateway but not being able to get to the BMC after changing one.

agangidi53, 2017-05-05:
Yup gateway finalizes the IP assignment as far as i remember:

Here's a snippet from my blog that works, in-band:

   ipmitool lan set 1 ipsrc static

   ipmitool lan set 1 ipaddr 10.127.xx.xx

   ipmitool lan set 1 netmask 255.255.xx.xx

   ipmitool lan set 1 defgw ipaddr 10.127.xx.xx

williamspatrick, 2017-05-05:
@agangidi53 - I'll warn you we might have changed that procedure in the master code stream.  There was some re-reading of the IPMI spec that happened and we didn't think it was appropriate to assume when the user was "done" making network changes.  Hopefully @tomjoseph83 remembers the details better than me.

tomjoseph83, 2017-05-05:
@agangidi53 @zet809 

After setting  the LAN fields (IP Address, Netmask, Gateway Address). It should be followed by the Set Channel Access command.

```ipmitool lan set 1 access on```

agangidi53, 2017-05-05:
Thanks @williamspatrick  @tomjoseph83 . I will test this on G2 with latest build.

zet809, 2017-05-08:
Hi, @tomjoseph83 , the 'set 1 access on' command doesn't work:
```
[xCAT Genesis running on p9euh02 /]# ipmitool lan set 1 ipaddr 10.6.9.200
Setting LAN IP Address to 10.6.9.200
[xCAT Genesis running on p9euh02 /]# ipmitool lan set 1 netmask 255.0.0.0
Setting LAN Subnet Mask to 255.0.0.0
[xCAT Genesis running on p9euh02 /]# ipmitool lan set 1 defgw ipaddr 10.0.0.101
Setting LAN Default Gateway IP to 10.0.0.101
[xCAT Genesis running on p9euh02 /]# ipmitool lan set 1 access on
IPMI command failed: Invalid command
Unable to Get Channel Access(non-volatile) for channel 1
```

tomjoseph83, 2017-05-08:
@zet809 Set Channel Access command triggers the applying of the IP settings. The ```ipmitool lan set 1 access on``` would issue Get Channel Access IPMI command to read the settings and then sent it over Set Channel Access command. The Get Channel access command is not currently supported.

As a workaround you can issue the Set Channel Access as a raw command.

```ipmitool raw 0x06 0x40```

I will add support for the Get Channel Access command so that you can achieve the same using ```ipmitool lan set 1 access on```

zet809, 2017-05-08:
hi, @tomjoseph83 , thx a lot.
In order to not forget the issue, will you pls reopen this defect and close it when the 'set access on' interface is ready?

zet809, 2017-06-20:
Hi, @tomjoseph83 , how is the interface implementing? May I know when will this interface ready? Thx!

zet809, 2017-07-06:
Hi, @tomjoseph83 , any update for this interface? Thx!

tomjoseph83, 2017-07-12:
@zet809 I will push a fix in the next week to support Get Channel Access command.

tomjoseph83, 2017-07-27:
- [ ] Implement Get Channel Access command

tomjoseph83, 2017-08-04:
I tested the changes using ipmitool over network (```ipmitool lan set 1 access on``` ) and this would be disruptive, since the IP changes and the response would not be sent back. The ipmitool would show the command did not succeed, but the network settings would get applied.

This issue would not be noticed when the command is executed from the host.

rfrandse, 2017-08-04:
<https://gerrit.openbmc-project.xyz/5774> Implement Get channel access command
Resolves  #1565 [xCAT]Set OpenBMC ip address with ipmitool in-band doesn't work

tomjoseph83, 2017-08-07:
# **STATUS**

8/7
Ready for merge review.

zet809, 2017-08-11:
Thx, @tomjoseph83

closed 2017-08-16 by williamspatrick

zet809, 2017-08-23:
Hi, @williamspatrick , is the interface ready to use? When will I get the firmware that have this interface opened? Thx!

whowutwut, 2017-08-28:
@zet809  On the GHE build 8/24, I see the following when doing inband from the Genesis Kernel. .
```
[xCAT Genesis running on (none) /]# ipmitool lan set 1 access on
IPMI command failed: Command response could not be provided
Unable to Set Channel Access(non-volatile) for channel 1
[xCAT Genesis running on (none) /]#
```

I did see a NCIS up event happen on the BMC, so I think it does get sent...  like @tomjoseph83  says above here: https://github.com/openbmc/openbmc/issues/1565#issuecomment-320212650 

Perhaps we should leave the RAW command and now switch back to the `access on`?