openbmc/openbmc #1566: [BarrelEye v1.99.*] If ESel hardware errors happened, then cannot boot host anymore and needs manual reupdate of PNOR

yadron, 2017-05-05:
How to reproduce:
1. Some hardware problem happened with registered ESel messages (there are many ways how to make a hardware problem).
2. After hard reboot, BMC booted.
3. Start booting host ("obmcutil poweron").
4. Start console ("obmc-util-console").
5. Result: cannot boot host, because frozen, both seeing such a messages as listed below (and in most cases nothing helps except of uploading new PNOR and only then rebooting).

Question: how to clear ESEL history from PNOR without updating complete PNOR image?

Advice: if such a froze expected (for example, so Admin can see it and needs cleaning it manually), then at least it should say something regarding this and how to clean before stalled with "ISTEP  6. 3 ". And better if it doesn't froze, but for example make a long delay and then continue trying booting host.

I've attached OpenBMC journalctl logs both for FAILed case and for NORMal host boot case:
[fail.txt](https://github.com/openbmc/openbmc/files/979918/fail.txt)
[norm.txt](https://github.com/openbmc/openbmc/files/979920/norm.txt)

```
root@barreleye:~# obmc-console-client
  2.44247|ERRL|Dumping errors reported prior to registration
  2.47943|================================================
  2.47943|Error reported by hwpf (0x0900)
  2.49789|  HW Procedure generated error. See User Data.
  2.49789|  ModuleId   0x02 MOD_HWP_RC_TO_ERRL
  2.49789|  ReasonCode 0x090f RC_HWP_GENERATED_ERROR
  2.49790|  UserData1  RC value from HWP : 0x00000000002bee06
  2.49790|  UserData2  <unused> : 0x0000000000000000
  2.49790|User Data Section 0, type UD
  2.49790|  Subsection type 0x06
  2.49791|  ComponentId errl (0x0100)
  2.49791|  CALLOUT
  2.49791|  HW CALLOUT
  2.49791|  Reporting CPU ID: 10
  2.49792|  Called out entity:
  2.49792|User Data Section 1, type UD
  2.49792|  Subsection type 0x06
  2.49792|  ComponentId errl (0x0100)
  2.49792|  CALLOUT
  2.49793|  CLOCK ERROR
  2.49793|  clockType: 2
  2.49793|User Data Section 2, type UD
  2.49793|  Subsection type 0x01
  2.49794|  ComponentId hwpf (0x0900)
  2.49794|  STRING
  2.49794|
  2.49794|User Data Section 3, type UD
  2.49795|  Subsection type 0x02
  2.49795|  ComponentId hwpf (0x0900)
  2.49795|  TARGET
  2.49795|  HUID: 000342c0
  2.49804|User Data Section 4, type UD
  2.49804|  Subsection type 0x02
  2.49804|  ComponentId hwpf (0x0900)
  2.49805|  TARGET
  2.49805|  HUID: 00000002
  2.49812|User Data Section 5, type UD
  2.49812|  Subsection type 0x04
  2.49812|  ComponentId errl (0x0100)
  2.49813|User Data Section 6, type UD
  2.49813|  Subsection type 0x02
  2.49813|  ComponentId errl (0x0100)
  2.49814|  TARGET
  2.49814|  HUID: 000d0018
  2.49820|  Phys path: Physical:/Sys0/Node0/Membuf24/MBA0
  2.49820|  Affinity path: Logical:/Sys0/Node0/Proc3/MCS0/Membuf24/MBA0
  2.49821|User Data Section 7, type UD
  2.49821|  Subsection type 0x03
  2.49821|  ComponentId errl (0x0100)
  2.49822|User Data Section 8, type UD
  2.49822|  Subsection type 0x01
  2.49822|  ComponentId errl (0x0100)
  2.49822|  STRING
  2.49823|  Hostboot Build ID:
  2.49823|User Data Section 9, type UD
  2.49823|  Subsection type 0x04
  2.49823|  ComponentId errl (0x0100)
  2.49824|User Data Section 10, type UD
  2.49824|  Subsection type 0x15
  2.49824|  ComponentId hb-trace (0x3100)
  2.49824|User Data Section 11, type UD
  2.49825|  Subsection type 0x15
  2.49825|  ComponentId hb-trace (0x3100)
  2.49826|User Data Section 12, type UD
  2.49826|  Subsection type 0x15
  2.49826|  ComponentId hb-trace (0x3100)
  2.49826|User Data Section 13, type UD
  2.49827|  Subsection type 0x15
  2.49827|  ComponentId hb-trace (0x3100)
  2.49827|================================================
  3.40930|================================================
  3.40930|Error reported by hwpf (0x0900)
  3.40931|  HW Procedure generated error. See User Data.
  3.40931|  ModuleId   0x02 MOD_HWP_RC_TO_ERRL
  3.40931|  ReasonCode 0x090f RC_HWP_GENERATED_ERROR
  3.40931|  UserData1  RC value from HWP : 0x00000000002bee06
  3.40932|  UserData2  <unused> : 0x0000000000000000
  3.40932|User Data Section 0, type UD
  3.40932|  Subsection type 0x06
  3.40932|  ComponentId errl (0x0100)
  3.40932|  CALLOUT
  3.40933|  HW CALLOUT
  3.40933|  Reporting CPU ID: 10
  3.40933|  Called out entity:
  3.40933|User Data Section 1, type UD
  3.40933|  Subsection type 0x06
  3.40934|  ComponentId errl (0x0100)
  3.40934|  CALLOUT
  3.40934|  CLOCK ERROR
  3.40934|  clockType: 2
  3.40934|User Data Section 2, type UD
  3.40935|  Subsection type 0x01
  3.40935|  ComponentId hwpf (0x0900)
  3.40935|  STRING
  3.40935|
  3.40935|User Data Section 3, type UD
  3.40936|  Subsection type 0x02
  3.40936|  ComponentId hwpf (0x0900)
  3.40936|  TARGET
  3.40936|  HUID: 00000002
  3.40945|User Data Section 4, type UD
  3.40945|  Subsection type 0x02
  3.40945|  ComponentId hwpf (0x0900)
  3.40946|  TARGET
  3.40946|  HUID: 01010900
  3.40953|User Data Section 5, type UD
  3.40953|  Subsection type 0x04
  3.40953|  ComponentId errl (0x0100)
  3.40953|User Data Section 6, type UD
  3.40954|  Subsection type 0x02
  3.40954|  ComponentId errl (0x0100)
  3.40954|  TARGET
  3.40954|  HUID: 000d0019
  3.40960|  Phys path: Physical:/Sys0/Node0/Membuf24/MBA1
  3.40961|  Affinity path: Logical:/Sys0/Node0/Proc3/MCS0/Membuf24/MBA1
  3.40961|User Data Section 7, type UD
  3.40962|  Subsection type 0x03
  3.40962|  ComponentId errl (0x0100)
  3.40962|User Data Section 8, type UD
  3.40962|  Subsection type 0x01
  3.40963|  ComponentId errl (0x0100)
  3.40963|  STRING
  3.40963|  Hostboot Build ID:
  3.40963|User Data Section 9, type UD
  3.40964|  Subsection type 0x04
  3.40964|  ComponentId errl (0x0100)
  3.40964|User Data Section 10, type UD
  3.40964|  Subsection type 0x15
  3.40965|  ComponentId hb-trace (0x3100)
  3.40965|User Data Section 11, type UD
  3.40965|  Subsection type 0x15
  3.40965|  ComponentId hb-trace (0x3100)
  3.40966|User Data Section 12, type UD
  3.40966|  Subsection type 0x15
  3.40966|  ComponentId hb-trace (0x3100)
  3.40967|User Data Section 13, type UD
  3.40967|  Subsection type 0x15
  3.40967|  ComponentId hb-trace (0x3100)
  3.40967|================================================
  3.46182|================================================
  3.46182|Error reported by ipmi (0x2500)
  3.48025|  Request to get power limit information failed
  3.48025|  ModuleId   0x04 IPMI::MOD_IPMIDCMI
  3.48025|  ReasonCode 0x250a IPMI::RC_DCMI_CMD_FAILED
  3.49877|  UserData1  BMC IPMI Completion code. : 0x00000000000000ce
  3.49877|  UserData2   : 0x0000000000000000
  3.49877|User Data Section 0, type UD
  3.49877|  Subsection type 0x06
  3.49878|  ComponentId errl (0x0100)
  3.49878|  CALLOUT
  3.49878|  PROCEDURE ERROR
  3.49878|  Procedure: 85
  3.49878|User Data Section 1, type UD
  3.49879|  Subsection type 0x03
  3.49879|  ComponentId errl (0x0100)
  3.49879|User Data Section 2, type UD
  3.49879|  Subsection type 0x01
  3.49880|  ComponentId errl (0x0100)
  3.49880|  STRING
  3.49880|  Hostboot Build ID:
  3.49880|User Data Section 3, type UD
  3.49881|  Subsection type 0x15
  3.49881|  ComponentId hb-trace (0x3100)
  3.49881|================================================
  4.41175|PNOR version : OS188PMSt17_OccEn_TLDis_Ibm100x200x_HostBoot3802_Reddragon0006_occ8001.
  4.43300|================================================
  4.43300|Error reported by htmgt (0x2600)
  4.43301|  <none>
  4.43301|  ModuleId   0x07 unknown
  4.43301|  ReasonCode 0x2616 unknown
  4.43301|  UserData1  unknown : 0x0000000000000000
  4.43302|  UserData2  unknown : 0x0000000000000000
  4.43302|User Data Section 0, type UD
  4.43302|  Subsection type 0x06
  4.43302|  ComponentId errl (0x0100)
  4.43303|  CALLOUT
  4.43303|  PROCEDURE ERROR
  4.43303|  Procedure: 16
  4.43303|User Data Section 1, type UD
  4.43303|  Subsection type 0x06
  4.43304|  ComponentId errl (0x0100)
  4.43304|  CALLOUT
  4.43304|  PROCEDURE ERROR
  4.43304|  Procedure: 85
  4.43305|User Data Section 2, type UD
  4.43305|  Subsection type 0x03
  4.43305|  ComponentId errl (0x0100)
  4.43305|User Data Section 3, type UD
  4.43306|  Subsection type 0x01
  4.43306|  ComponentId errl (0x0100)
  4.43306|  STRING
  4.43306|  Hostboot Build ID:
  4.43307|User Data Section 4, type UD
  4.43307|  Subsection type 0x15
  4.43307|  ComponentId hb-trace (0x3100)
  4.43308|================================================
  4.49782|ISTEP  6. 3 
```

williamspatrick, 2017-05-05:
There is no esel history in PNOR maintained by the BMC.  If the host firmware is reporting errors you'll have to take this up with them to get some decoding.  open-power/hostboot is the firmware reporting this problem and halting.

The PNOR version string here makes it look like you have custom host firmware on this machine since that is not a valid open-source version.  You may need to supply them with a summary of the changes you've made to the host firmware, as one of those changes might be causing this behavior.

closed 2017-05-05 by williamspatrick

yadron, 2017-05-05:
Actually the problem is that the reason of hardware fail already fixed, but still cannot boot host until PNOR updated. So for sure it caches previous ESel messages someone in PNOR or somewhere else and both some-kind of flag in PNOR. Where it caches the prev. ESel messages and why updating PNOR image helps erasing it history?

yadron, 2017-05-30:
Found the way how to fix it (so yes, some ESel or related history stored in PNOR):

pflash -c -P HBEL --direct

pflash -c -P DJVPD --direct

williamspatrick, 2017-05-30:
Both of those mentioned PNOR partitions (HBEL, DJVPD) are entirely under the control of the host firmwrae (Hostboot).  The BMC is not involved in any caching or direct modification of them.