openbmc/openbmc #1567: Pass arguments to phosphor-fan-control

spinler, 2017-05-05:
We're going to add arguments to phosphor-fan-control to be able to do 2 things:

1)  On the boot, use phosphor-fan-control --init(or something) to
   - set fans to max speed (as defined in some yaml)
   - sleep for 20s (defined in yaml)  allowing the fans to get to high speed

2) phosphor-fan-control --control (or something) to actually start the fan control loop

then after 1) runs, we can start phosphor-fan-monitor and phosphor-fan-control --control, and they can both expect the fans to be up to speed

We still need to investigate some on if 1) is needed on a reset at runtime.

We're also going to have phosphor-fan-presence run starting at standby so it won't be involved in any race conditions with the others, and also GPIOs are accessible at standby

spinler, 2017-05-05:
- [ ] Add arguments to phosphor-fan-control as described above
- [ ] Have service file that calls the --init on boot start, followed by --control and fan-monitor

rfrandse, 2017-05-18:
https://gerrit.openbmc-project.xyz/4234 Make use of obmc-fan-control-ready@.target
Resolves #1567 Pass arguments to phosphor-fan-control

closed 2017-05-23 by williamspatrick