openbmc/openbmc #1568: Output early host boot progress messages to the host console

rlippert, 2017-05-05:
In the interest of only having to watch one console output to see what is going on with a machine, I think it would be useful for the BMC to output the state of the poweron/boot progress to the host console (e.g. the thing readable via obmc-console-client).

For example each of these events would be useful to see something printed out on the obmc-console-client:

* CPU power being turned on or off
* host watchdog failures
* host checkstop detected
* SBE being started
* SBE boot progress (before SBE reaches step 3)

The current approach of running "poweron" and then waiting for 30s for something to hopefully output to the console is IMO not a very good user experience.

@GHF

KennethWilke, 2017-05-05:
:+1: I think the addition of a message written when the poweron and poweroff commands are executed would be helpful as well

agangidi53, 2017-05-05:
@KennethWilke  @rlippert  This would be good to have. But doubt if the user will be able to those very early messages , especially if the they are connected on a network via obmc-console-client. On both Barreleye G1 and G2, there's a link flap to BMC interfaces that occurs for ~ 2 seconds. It's more consistent on power off . Also happens during power on, albeit a bit inconsistently.

williamspatrick, 2017-05-06:
@agangidi53 - someone should investigate why your G2 machine sees that. We do not on Witherspoon (also NCSI). G1 used a NCSI chip with a known issue in this area.

gtmills, 2018-05-31:
@rlippert  @agangidi53 Still a problem? Can this be closed? No update in over a year.

KennethWilke, 2018-06-01:
I don't believe this kind of functionality has been added, would still be useful

rlippert, 2018-06-01:
Yes still would be nice especially for openpower systems that take forever to start booting.

Might make sense to break the various requests out into separate bugs to make it more manageable?

closed 2019-04-27 by stale[bot]