openbmc/openbmc #1570: How to active RMCP+ session?

evan228, 2017-05-08:
I build a version for ast2500.
I want active RMCP session, but it shw error.
$ ipmitool -U root -P 0penBmc -H 192.168.1.8 -I lanplus mc info
Error: Unable to establish IPMI v2 / RMCP+ session
$ ipmitool -U root -P 0penBmc -H 192.168.1.8 -I lan mc info
Authentication type NONE not supported
Error: Unable to establish LAN session
Error: Unable to establish IPMI v1.5 / RMCP session

If i active netipmid, 
root@evb-ast2500:~# netipmid 
I> Registering Command100000
I> Registering Command120000
I> Registering Command140000
I> Registering Command1838
I> Registering Command183b
I> Registering Command183c
I> Registering Command10000
I> Registering Command1848
I> Registering Command1849
I> Registering Command184a

How can I build RMCP seeion?
Thanks

tomjoseph83, 2017-05-08:
```ipmitool -P 0penBmc -H 192.168.1.8 -I lanplus mc info```

The default user account is nameless, so no need to mention the username.

evan228, 2017-05-09:
Thank you, I have another question. I can't see the RMCP log. How to show the RMCP seeion log? Thanks

closed 2017-05-11 by evan228