openbmc/openbmc #1571: [Witherspoon] System is unable to recover from Quiesced state after first host off request.

rahulmah, 2017-05-08:
Witherspoon system is unable to recover from Quiesce state after first host off request. This issue is an inconsistent one.

Step to reproduce
1. Get system is quiesced state. 
2. Now send REST request to power off the host to recover from Quiesced state. Here you will see that system will not able to recover from Quiesced state.  Waited for almost 15 mins.

```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.State.Host.Transition.Off\"}" -X PUT  https://xx.xx.xx.xx/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
```
3. Send REST Request to power off host again. Here we will see that host will go into off state now.

gkeishin, 2017-05-22:
Couple of things we need to ensure for this @rahulmah 
 
- [x]  Which PNOR are you using . PNOR level please ?
- [x]  Did you mean we can't poweroff once quisece ? what was the auto boot policy at that time..

rahulmah, 2018-05-24:
Closing this issue as we dint observe the problem later on.

closed 2018-05-24 by rahulmah