openbmc/openbmc #1574: Implement host factory reset

geissonator, 2017-05-08:
Implement factory reset interface for host.  This will mostly be just clearing PNOR areas.  Need to work with host team on details of where to clear.  Then work with @anoo1 to get required interface(s) on BMC side to do the required operations on PNOR.

mtritz, 2017-05-09:
- [ ] Figure out where to put the new object
- [ ] Create a skeleton to implement the dbus interface
- [ ] Identify which PNOR partitions to clear
- [ ] Identify the BMC interface(s) needed to clear those partitions
- [ ] Add to skeleton - when dbus interface is called, clear the PNOR partitions

mtritz, 2017-05-15:
Clear partitions with a reprovision flag.

rfrandse, 2017-05-18:
https://gerrit.openbmc-project.xyz/4253 Implement host factory reset in item_updater
Resolves #1574 Implement host factory reset

closed 2017-05-24 by williamspatrick