openbmc/openbmc #1575: Implement network factory reset

geissonator, 2017-05-08:
Implement factory reset dbus interface in network object.  The interface should reset all network interfaces to defaults (i.e. dhcp)

mtritz, 2017-05-17:
- [ ] Figure out where to put the new object
- [ ] Create a skeleton to implement the dbus interface

- [ ] TODO: Identify what exactly a "network factory reset" encompasses and how to do it. (Design meeting)

- [ ] Add to skeleton - when dbus interface is called, reset network interfaces

ratagupt, 2017-05-19:
we just  need to delete the /etc/systemd/network/*.network file.

rfrandse, 2017-05-23:
https://gerrit.openbmc-project.xyz/4321 Implement network factory reset in network_manager
Resolves #1575 Implement network factory reset

closed 2017-06-05 by williamspatrick