openbmc/openbmc #1576: Implement BMC factory reset

geissonator, 2017-05-08:
Still need to work out the details here.  Will be implemented based on BMC code update design and persistent data designs.

mtritz, 2017-06-28:
- [ ] Create a service file to delete the BMC read-write volumes
- [ ] Identify where to put the BMC reset dbus object
- [ ] Implement the factory reset interface in that location
- [ ] Write a function that calls the service file from step 1 when the BMC reset interface is called

mdmillerii, 2017-06-28:
We need to explicitly sanitize the u-boot env to have just the inventory mac and uuid from the default.
The read-write volume should probably be reset vs deleted, especially as we go to ubifs and use a fixed name.

mtritz, 2017-07-13:
**Daily status:**
**7/13:** We held design discussions about what exactly the service files needed to do. Rather than simply deleting the read-write volumes, upon the presence of an environment variable "rwreset=true" on boot, a new service file will call the obmc-flash-bmc script to remove and recreate the read-write volume. Accordingly, I created two new service files and extended that script.

rfrandse, 2017-07-25:
https://gerrit.openbmc-project.xyz/#/c/5244/ 
and 
https://gerrit.openbmc-project.xyz/#/c/5302/
resolve #1576 according to Michael

closed 2017-07-25 by rfrandse