openbmc/openbmc #1578: [Witherspoon] Unable to set power cap using IPMI

rahulmah, 2017-05-09:
Unable to set power cap using IPMI.

```
ipmitool -I lanplus -C 3 -P 0penBmc -H xx.xx.xx.xx raw 0x04 0x30 0xC2 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x20 0x00
```

BMC Fw: 
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-19"
VERSION_ID="v1.99.5-19-g02406c9"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-19"
BUILD_ID="v1.99.5"
```

gkeishin, 2017-05-09:
@rahulmah  Can u add the snapshot log for before and after when you do this operation and also the journald log to see if the IPMI messages are coming through...

rahulmah, 2017-05-09:
Setting power cap using IPMI is disabled at present. (https://gerrit.openbmc-project.xyz/#/c/3742/.) So closing this.

closed 2017-05-09 by rahulmah

sivassrr, 2017-06-06:
Is it enabled now, can we revisit this.

rahulmah, 2017-06-06:
@sivassrr : As discussed dev as disabled it for time being. It will be revisited when once it is enabled.