openbmc/openbmc #1579: [Barreleye] Barreleye code update is stuck stuck during code update.

rahulmah, 2017-05-09:
In Barreleye, we saw one instance where code update with latest build 732 is stuck at below point in serial console.

```
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
+ umount /mnt/initramfs/ro
+ umount /mnt
+ set +x
update: --clean-saved-files
Updating bmc...
Erasing block: 366/512 (71%) 
```

Full Serial Console Log:
[serial console logs for barrleye code update failure.txt](https://github.com/openbmc/openbmc/files/986282/serial.console.logs.for.barrleye.code.update.failure.txt)

geissonator, 2017-05-09:
@rahulmah is this the system where you said you accidentally flashed a witherspoon image?  Not sure why that would affect the erase but you never know.  Otherwise this looks like a dup of #522

rahulmah, 2017-05-09:
Andrew: No, this is not the system where I accidentally flashed incorrect image. 
And yes it seems like dup of #522 issue.

mdmillerii, 2017-05-09:
Was the system previously on 1.99.1 or later?  Or was it previously on a prior release, which would have the issue?

Also, if we add the `watchdog` command to the `busybox` config we can poke and or adjust the watchdog during the flash update.   The  70% done in 10 minutes is only the erase, the program (write) takes some additional time but typically less than 10% of the erase time.  The read is again 1-2 orders of magnitude faster.

rahulmah, 2018-05-24:
Closing this issue as we dint observe the problem later on.

closed 2018-05-24 by rahulmah