openbmc/openbmc #1581: std::unique_ptr not working with global in dynamic library

geissonator, 2017-05-09:
See https://gerrit.openbmc-project.xyz/#/c/3369/11/host-services.cpp for the issue.  

Basically when I use host as a unique_ptr, I can put things on the queue within the object fine when coming in via the DBUS interface, but when the same code in that source file tries to use the host object, it gets seg faults whenever it uses any of the members in the object (it can call the getNextCmd() function against the object but any member access in there seg faults).

mine260309, 2017-05-12:
I do not believe `std::unique_ptr` has problem with dynamic library, so I did some investigation.
If you check the `sdbus` and `host` when issue occurs, you can see both are `NULL`.

Then it's found that if you change the lines
```
std::unique_ptr<sdbusplus::bus::bus> sdbus = nullptr;
std::unique_ptr<phosphor::host::Host> host = nullptr;
```
to
```
std::unique_ptr<sdbusplus::bus::bus> sdbus;
std::unique_ptr<phosphor::host::Host> host;
```
Then it works fine.

I have no idea what the root cause is yet, it seems to indicate that `= nullptr` causes the variables to be initialized multiple times?

mine260309, 2017-05-15:
I asked about this at SO: http://stackoverflow.com/questions/43941159/global-static-variables-initialization-issue-with-attribute-constructor-i

It turns out to be related to a GCC bug, or more precisely, unspecified behavior, in short, the `std::unique_ptr` is initialized after `constructor`.

This issue can be fixed by adding `__attribute__((init_priority(101)))` to the global/static variables in the shared libraries, this makes sure they're initialized before `constructor`.
The priority number can be any number between `101` ~ `65534`, where `0` ~ `100` is reserved, and `65535` is the default priority for `constructor`.

mine260309, 2017-05-15:
Review: https://gerrit.openbmc-project.xyz/#/c/4212/

rfrandse, 2017-05-16:
https://gerrit.openbmc-project.xyz/4212 Set init_priority attribute for global/static variables
Resolves #1581 std::unique_ptr not working with global in dynamic library

closed 2017-05-19 by williamspatrick