openbmc/openbmc #1582: ConditionFileExists=/var/lib/obmc/cfam_overrides not valid in cfam_override@.service

geissonator, 2017-05-10:
Don't think there's any such thing as ConditionFileExists.

Probably want ConditionFileNotEmpty=
or
ConditionPathExists=

Saw this error when doing some other testing:
```
May 01 18:39:20 witherspoon systemd[1]: [/lib/systemd/system/obmc-chassis-poweron@0.target.requires/../cfam_override@.service:11] Unknown lvalue 'ConditionFileExists' in section 'Unit'
```

mtritz, 2017-05-10:
- [ ] Change ConditionFileExists to ConditionFileNotEmpty

rfrandse, 2017-05-11:
https://gerrit.openbmc-project.xyz/4164 cfam_override@.service - run on ConditionPathExists
Resolves #1582 ConditionFileExists=/var/lib/obmc/cfam_overrides not valid in cfam_override@.service

closed 2017-05-12 by williamspatrick