openbmc/openbmc #1583: I always get "Get Device ID command failed".

evan228, 2017-05-10:
Why I always get "Get Device ID command failed"?
Please help me.
Thanks

$ ipmitool  -P 0penBmc -H 192.168.1.1 -I lanplus raw 0x06 0x2
Get Device ID command failed: 0xc1 Invalid command
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x2 rsp=0xc1): Invalid command

==============================================================================
May 04 05:03:58 evb-ast2500 netipmid[787]: >> GetChannelCapabilities
May 04 05:03:58 evb-ast2500 netipmid[787]: << GetChannelCapabilities
May 04 05:03:58 evb-ast2500 netipmid[787]: >> openSession
May 04 05:03:58 evb-ast2500 netipmid[787]: << openSession
May 04 05:03:58 evb-ast2500 netipmid[787]: >> RAKP12
May 04 05:03:58 evb-ast2500 netipmid[787]: Failed to get /org/openbmc/control/chassis0 bus name: No such file or directory
May 04 05:03:58 evb-ast2500 netipmid[787]: << RAKP12
May 04 05:03:58 evb-ast2500 netipmid[787]: >> RAKP34
May 04 05:03:58 evb-ast2500 netipmid[787]: Failed to get /org/openbmc/control/chassis0 bus name: No such file or directory
May 04 05:03:58 evb-ast2500 netipmid[787]: << RAKP34
May 04 05:03:58 evb-ast2500 netipmid[787]: >> setSessionPrivilegeLevel
May 04 05:03:58 evb-ast2500 netipmid[787]: << setSessionPrivilegeLevel
May 04 05:03:58 evb-ast2500 netipmid[787]: E> Table:: Command Not found: 0xb03e
May 04 05:03:58 evb-ast2500 netipmid[787]: > Table:: Command Not found: 0x1801
May 04 05:03:58 evb-ast2500 netipmid[787]: > Table:: Command Not found: 0xb000
May 04 05:03:58 evb-ast2500 netipmid[787]: > Table:: Command Not found: 0x1801

==============================================================================
root@evb-ast2500:/etc# cat os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-dirty"
VERSION_ID="v1.99.5-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-dirty"
BUILD_ID="v1.99.5"

tomjoseph83, 2017-05-11:
@evan228 

From where are you running the ipmitool? 192.168.1.1 is the loopback address, if you are running from a machine outside OpenBMC, you need to specify the IP address of the OpenBMC.

```ipmitool -P 0penBmc -H 192.168.1.1 -I lanplus raw 0x06 0x2```

evan228, 2017-05-11:
Sorry, I changed this IP. Please check the Log , this is not the IP error by the issue.

tomjoseph83, 2017-05-11:
@evan228 The Get Device ID is Command 0x01 and not 0x02.

```
-bash-4.1$ ipmitool -I lanplus -C 3 -P 0penBmc -H 9.3.185.164 raw 0x06 0x01
 00 80 01 99 02 8d 41 a7 00 4f 42 00 05 01 00 ```

evan228, 2017-05-11:
Yes. The command is 0x02, but always get "Get Device ID command failed". I have tried other command, but I always get "Get Device ID command failed. I have find the ast2500 image is no "ipmid". Is this the Root Cause?

tomjoseph83, 2017-05-11:
I do not understand your question here. Are you trying to highlight the problem with "Get Device ID" command? Try to restart the net-ipmid service and try again. If you hit the problem, share the journalctl output.

evan228, 2017-05-12:
This is my command. I try the different command, the output is "Get Device ID command failed: 0xc1 Invalid command".
[image]

The following is journalctl output.
[https://www.dropbox.com/s/rieudh83oeykxaa/ast2500_ipmi.txt?dl=0](url)
Thanks

lucient168, 2017-07-24:
Is any updated, I also hit this issue?

tomjoseph83, 2017-08-07:
@lucient168 Please share the journal output after restarting the phosphor-ipmi-net. ```systemctl restart phosphor-ipmi-net```

@evan228 Are you still hitting this issue?

henbin, 2017-08-08:
Hi,
I met the same symptom

[image]

Could you help me?
Thanks.

[log.zip](https://github.com/openbmc/openbmc/files/1207691/log.zip)

tomjoseph83, 2017-08-08:
@henbin I do not see any IPMI traces in the logs you attached. Can you take the traces when you run this command, ```systemctl restart phosphor-ipmi-net```

henbin, 2017-08-09:
Hi Tom,
I attached the console log for your reference. Please tell what trace you need if they aren't.

[evb-2500.zip](https://github.com/openbmc/openbmc/files/1209912/evb-2500.zip)
Thanks.

tomjoseph83, 2017-08-10:
@henbin There are no net-ipmid start-up traces, please take the journal output simultaneously when the service is restarted. This could have happened because of the symbol mismatch with the IPMI provider libarary( fixed with #2065 ). I was looking for the traces to confirm the behaviour.

This has happened a few times earlier, i hope that is why it is reported multiple times. With the current master, you shouldn't be hitting this.

@evan228 @lucient168

henbin, 2017-08-10:
@tomjoseph83  I attach the journal output for your reference. I will pull the latest master and rebuild the evb-ast2500.

Thanks.
[journal_out.zip](https://github.com/openbmc/openbmc/files/1214584/journal_out.zip)

henbin, 2017-08-14:
Hi Tom,
I confirmed I have the latest commit for the phosphor-ipmi-net REPO.
[image]

I find the below commands (APP NetFun) are implemented only but not include 'Get Device ID Command'. 
That may be why 'Get Device ID command failed'

[image]

Could you give some suggestion?
Thanks.

tomjoseph83, 2017-08-16:
@henbin The above mentioned commands are the one that are network IPMI specific. The remaining standard commands are provided by the IPMI provider library in the phosphor-host-ipmid repository.

henbin, 2017-08-17:
@tomjoseph83 I built the evb-ast2500 target 
expoer TEMPLATECONF=meta-openbmc-machines/meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf

And ran the built FW on the physical AST2400 EVB.
I didn't see those directories you mentioned.

[image]

Thanks.

henbin, 2017-08-18:
@tomjoseph83 I could send Get Device ID command after I add the define of OBMC_MACHINE_FEATURES in the evb-ast2500.conf. But I met another problems as the below,
[image]
**All response data is zero.**
And I checked the journal of phosphor-ipmi-net and found "Failed to get /org/openbmc/inventory/system/chassis/motherboard/bmc"
[image]

I tried the REST function and got the same symptom.
[image]

May I lose some setting? Do you have any suggestion? 
Thanks.

tomjoseph83, 2017-08-18:
@henbin The /org/openbmc/inventory/system/chassis/motherboard/bmc is not cretaed until the host sends down information. So this is expected, since you have an evaluation board and there is no host.

@evan228 @lucient168 Please try to make changes to *.conf file and give a try.

henbin, 2017-08-18:
@tomjoseph83 Thanks for your explanation. But I don't understand why the obtainment of BMC revision is from host.

tomjoseph83, 2017-08-18:
@henbin Probably i am wrong on that one. I am not sure how the inventory is populated in the case of evaluation board, that would need some investigation.

dkodihal, 2017-08-21:
Looks like the original question/concern in this defect has been responded to by @tomjoseph83, so closing this.

closed 2017-08-21 by dkodihal