openbmc/openbmc #1584: pwmX_enable hwmon sysfs file needs to be written

spinler, 2017-05-10:
When doing testing after I removed the set-default-fans-speed.sh script and service, I found that after AC is pulled, the fans cannot be programmed by hwmon at all.

This is because that script was doing some register inits to the MAX chip to enable fan control, which can also be accomplished  by writing /sys/class/hwmon/hwmon1/pwmX_enable, where X is the fan number.

```pwm[1-6]_enable RW pwm mode, 0=disabled, 1=pwm, 2=rpm, 3=automatic```

spinler, 2017-05-10:
We're going to have phosphor-hwmon do this somehow.

rfrandse, 2017-05-11:
https://gerrit.openbmc-project.xyz/4176 Write pwmX_enable for fans that have a fanX_target
Resolves #1584 pwmX_enable hwmon sysfs file needs to be written

closed 2017-05-12 by williamspatrick