openbmc/openbmc #1593: Soft power off by pressing power button broken

mine260309, 2017-05-11:
When host is powered on, pressing power button should soft-power-off the host.

This function is broken because `chassis_control.py` is using the legacy `org.openbmc.HostServices`, which does not exist.
```
Apr 19 08:37:53 lanyang chassis_control.py[813]: Soft off power
Apr 19 08:37:53 lanyang chassis_control.py[813]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Apr 19 08:37:53 lanyang chassis_control.py[813]: Traceback (most recent call last):
Apr 19 08:37:53 lanyang chassis_control.py[813]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Apr 19 08:37:53 lanyang chassis_control.py[813]:     self._handler(*args, **kwargs)
Apr 19 08:37:53 lanyang chassis_control.py[813]:   File "/usr/sbin/chassis_control.py", line 184, in power_button_signal_handler
Apr 19 08:37:53 lanyang chassis_control.py[813]:     self.softPowerOff()
......
Apr 19 08:37:53 lanyang chassis_control.py[813]: DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.HostServices was not provided by any .service files
```

mine260309, 2017-05-11:
Either fix `chassis_control.py` or wait it to be replaced/refactored by some `xyz.openbmc_project.xxx` service.

geissonator, 2017-07-29:
- [ ] Move all button press code to using new host and chassis interfaces
- [ ] Verify all button presses work as expected
- [ ] Ensure test team has adequate coverage of button presses

# **STATUS**
7/29 - No progress yet

rfrandse, 2017-07-30:
<https://gerrit.openbmc-project.xyz/5641> Update button press handlers to new targets
Resolves  #1593 Soft power off by pressing power button broken

closed 2017-08-15 by williamspatrick