openbmc/openbmc #1595: Software BMC Version service keeps restarting [Witherspoon]

gkeishin, 2017-05-11:
This is new builds, this behavior started showing up and pervasive in newer master builds 

```
Apr 20 17:31:02 witherspoon systemd[1]: Failed to start Phosphor Version Software Manager.
Apr 20 17:31:02 witherspoon systemd[1]: xyz.openbmc_project.Software.BMC.Version.service: Unit entered failed state.
Apr 20 17:31:02 witherspoon systemd[1]: xyz.openbmc_project.Software.BMC.Version.service: Failed with result 'timeout'.
```

`Driver info`
```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/736/
- [x] VERSION_ID : v1.99.5-26-geb2af39
- [x] Kernel     : Linux witherspoon 4.7.10-15c30d4deb0a01f398230395dc0d9af4be92d856 #1 Thu May 11 04:21:21 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.16_1.3
        op-build-v1.16-35-g2bb0a83-dirty
        buildroot-2017.02-4-gc28573e
        skiboot-5.5.0
        hostboot-c9bd42f
        linux-4.10.7-openpower1-p61306aa
        petitboot-v1.4.2-p58d71a5
        witherspoon-xml-3a7d05c
        occ-e1889e5
        hostboot-binaries-8cbbc67
        capp-ucode-1bb7503
        sbe-9b98de3-p554c410
```

- [x] Journald log 
[journald_v1.99.5-26-geb2af39_Software.BMC.Version.service.txt](https://github.com/openbmc/openbmc/files/992464/journald_v1.99.5-26-geb2af39_Software.BMC.Version.service.txt)

williamspatrick, 2017-05-11:
Likely means a dbus object defined in its service file is no longer being created. Was there a change to the registered service name @anoo1?

anoo1, 2017-05-11:
There was, let me track down the change to see if there's anything pending still in review.

anoo1, 2017-05-11:
@gtmills will upload the change to rename the service file. The service file has been renamed a couple times between BMC and Version so we just need one more change for this last update.

gtmills, 2017-05-11:
https://gerrit.openbmc-project.xyz/#/c/4172/

closed 2017-05-11 by williamspatrick