openbmc/openbmc #1596: Host state is shown as Quiesced though Host Linux is up and running

trajeswaran, 2017-05-11:
I am able to login to Host Linux. But, the Rest API /xyz/openbmc_project/state/host0/attr/CurrentHostState  shows it as Quiesced.

/xyz/openbmc_project/state/host0/attr/CurrentHostState       
{
  "data": "xyz.openbmc_project.State.Host.HostState.Quiesced", 
  "message": "200 OK", 
  "status": "ok"

Please find attached the FFDC:
My FFDC Logs :: This test is needed to satisfy FFDC initial setup ... | PASS |
------------------------------------------------------------------------------
Myffdc :: This module is for triggering data collection on demand     | PASS |
1 critical test, 1 passed, 0 failed
1 test total, 1 passed, 0 failed
==============================================================================
Output:  /gsa/ausgsa-p10/03/indiateam04/rajeswgo/projects/openbmc-test-automation/output.xml
Log:     /gsa/ausgsa-p10/03/indiateam04/rajeswgo/projects/openbmc-test-automation/log.html
Report:  /gsa/ausgsa-p10/03/indiateam04/rajeswgo/projects/openbmc-test-automation/report.html
[rajeswgo@gfwa124 openbmc-test-automation]$ ll /gsa/ausgsa-p10/03/indiateam04/rajeswgo/projects/openbmc-test-automation/logs/20170511061127600612_Myffdc/20170511061127600612_MyFFDCLogs/
total 640
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin  12428 May 11 06:11 20170511061127600612_BMC_dmesg.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin  16121 May 11 06:11 20170511061127600612_BMC_elog.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin   3538 May 11 06:11 20170511061127600612_BMC_general.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin  15027 May 11 06:11 20170511061127600612_BMC_inventory.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin 576807 May 11 06:11 20170511061127600612_BMC_journalctl.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin     65 May 11 06:11 20170511061127600612_BMC_led.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin    952 May 11 06:11 20170511061127600612_BMC_meminfo.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin    291 May 11 06:11 20170511061127600612_BMC_procinfo.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin   8699 May 11 06:11 20170511061127600612_BMC_proc_list.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin   1363 May 11 06:11 20170511061127600612_BMC_sensor_list.txt
-rw-rw-r-- 1 rajeswgo p/indiateam04/admin  10639 May 11 06:11 20170511061127600612_esel
[20170511061127600612_BMC_dmesg.txt](https://github.com/openbmc/openbmc/files/992801/20170511061127600612_BMC_dmesg.txt)
[20170511061127600612_BMC_elog.txt](https://github.com/openbmc/openbmc/files/992802/20170511061127600612_BMC_elog.txt)
[20170511061127600612_BMC_general.txt](https://github.com/openbmc/openbmc/files/992804/20170511061127600612_BMC_general.txt)
[20170511061127600612_BMC_inventory.txt](https://github.com/openbmc/openbmc/files/992803/20170511061127600612_BMC_inventory.txt)
[20170511061127600612_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/992807/20170511061127600612_BMC_journalctl.txt)
[20170511061127600612_BMC_led.txt](https://github.com/openbmc/openbmc/files/992806/20170511061127600612_BMC_led.txt)
[20170511061127600612_BMC_meminfo.txt](https://github.com/openbmc/openbmc/files/992805/20170511061127600612_BMC_meminfo.txt)
[20170511061127600612_BMC_procinfo.txt](https://github.com/openbmc/openbmc/files/992808/20170511061127600612_BMC_procinfo.txt)
[20170511061127600612_BMC_proc_list.txt](https://github.com/openbmc/openbmc/files/992810/20170511061127600612_BMC_proc_list.txt)
[20170511061127600612_BMC_sensor_list.txt](https://github.com/openbmc/openbmc/files/992809/20170511061127600612_BMC_sensor_list.txt)

gkeishin, 2017-05-11:
I think we all are chasing the same thing On P8 boxes Barreleye and palmetto..https://github.com/openbmc/openbmc/issues/1426

Now we are seeing in witherspoon p9 boxes too.

geissonator, 2017-05-11:
This one is the normal one where SBE/Hostboot took longer then the allotted 30s to boot up and reset the watchdog.  In this case they were 3s too late.  
```
Apr 22 09:24:25 witherspoon systemd[1]: Starting Start host0 SBE...
Apr 22 09:24:25 witherspoon openpower-proc-control[2076]: Running P9 procedure startHost
Apr 22 09:24:25 witherspoon systemd[1]: Started Start host0 SBE.
Apr 22 09:24:25 witherspoon phosphor-chassis-state-manager[874]: Received signal that power ON is complete
Apr 22 09:24:25 witherspoon phosphor-chassis-state-manager[874]: Change to Chassis Power State
Apr 22 09:24:25 witherspoon host_watchdog.exe[780]: Setting watchdog poll interval to: 30000
Apr 22 09:24:25 witherspoon host_watchdog.exe[780]: Starting watchdog with poll interval: 30000
Apr 22 09:24:25 witherspoon systemd[1]: Reached target Host0 (Starting).
Apr 22 09:24:25 witherspoon systemd[1]: Reached target Host0 (Started).
Apr 22 09:24:25 witherspoon systemd[1]: Starting Start Watchdog0...
Apr 22 09:24:25 witherspoon systemd[1]: Started Start Watchdog0.
Apr 22 09:24:25 witherspoon systemd[1]: Reached target Start Host0.
Apr 22 09:24:25 witherspoon phosphor-host-state-manager[868]: Received signal that host is running
Apr 22 09:24:25 witherspoon phosphor-host-state-manager[868]: Change to Host State
Apr 22 09:24:55 witherspoon host_watchdog.exe[780]: Polling watchdog: 0
Apr 22 09:24:55 witherspoon chassis_control.py[773]: Watchdog Error, Going to quiesce
Apr 22 09:24:55 witherspoon systemd[1]: Reached target Quiesce Target.
Apr 22 09:24:55 witherspoon phosphor-host-state-manager[868]: Auto reboot disabled. Maintaining quiesce.
Apr 22 09:24:55 witherspoon phosphor-host-state-manager[868]: Change to Host State
Apr 22 09:24:58 witherspoon host_watchdog.exe[780]: Setting watchdog poll interval to: 120000
Apr 22 09:24:58 witherspoon host_watchdog.exe[780]: Stopping watchdog
Apr 22 09:24:58 witherspoon host_watchdog.exe[780]: Starting watchdog with poll interval: 120000
```

geissonator, 2017-05-11:
Please open defect on sbe team.  They'll need PNOR level and any SOL console data available.

closed 2017-05-11 by geissonator