openbmc/openbmc #1597: WARN in ncsi_start_channel_monitor code at boot

gkeishin, 2017-05-11:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-28"
VERSION_ID="v1.99.5-28-g60b4452"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-28"
BUILD_ID="v1.99.5"
root@witherspoon:~# uname -a
Linux witherspoon 4.7.10-15c30d4deb0a01f398230395dc0d9af4be92d856 #1 Thu May 11 07:21:01 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

dmsg log
```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 847 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:240 ncsi_start_channel_monitor+0x40/0x78
CPU: 0 PID: 847 Comm: kworker/0:6 Not tainted 4.7.10-15c30d4deb0a01f398230395dc0d9af4be92d856 #1
Hardware name: ASpeed SoC
Workqueue: events ncsi_dev_work
[<8010723c>] (unwind_backtrace) from [<801054c4>] (show_stack+0x10/0x14)
[<801054c4>] (show_stack) from [<80110be0>] (__warn+0xd8/0xf4)
[<80110be0>] (__warn) from [<80110cf0>] (warn_slowpath_null+0x1c/0x24)
[<80110cf0>] (warn_slowpath_null) from [<80435a34>] (ncsi_start_channel_monitor+0x40/0x78)
[<80435a34>] (ncsi_start_channel_monitor) from [<804366c0>] (ncsi_configure_channel+0x294/0x2d4)
[<804366c0>] (ncsi_configure_channel) from [<80436e88>] (ncsi_dev_work+0x3a8/0x3d8)
[<80436e88>] (ncsi_dev_work) from [<80122008>] (process_one_work+0x18c/0x2f0)
[<80122008>] (process_one_work) from [<8012241c>] (worker_thread+0x280/0x3cc)
[<8012241c>] (worker_thread) from [<801267ec>] (kthread+0xdc/0xf0)
[<801267ec>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
---[ end trace c8b731184dbdbd8e ]---
```

gkeishin, 2017-05-11:
old bug reference : https://github.com/openbmc/openbmc/issues/564

gkeishin, 2017-05-11:
- [x] Journald log 
[journald_v1.99.5-28-g60b4452_IPL_failure_stack_trace.txt](https://github.com/openbmc/openbmc/files/992903/journald_v1.99.5-28-g60b4452_IPL_failure_stack_trace.txt)

- [x] dmsg log  
[journald_v1.99.5-28-g60b4452_IPL_failure_dmsg.txt](https://github.com/openbmc/openbmc/files/992911/journald_v1.99.5-28-g60b4452_IPL_failure_dmsg.txt)

shenki, 2017-05-12:
@gkeishin how did you trigger this behaviour?

How often does it occur?

gkeishin, 2017-05-12:
I saw it couple of time but need to see in today's new build. @shenki

gkeishin, 2017-05-12:
On the Master driver test https://openpower.xyz/job/openbmc-build/740/ build, I have not seen it today so far but on kernel 4.10 build https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/3547/   I ll see this

```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 9459 at /var/lib/jenkins-slave/workspace/openbmc-build-gerrit-trigger-multi/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:240 ncsi_start_channel_monitor+0x40/0x78
CPU: 0 PID: 9459 Comm: kworker/0:3 Not tainted 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1
Hardware name: ASpeed SoC
Workqueue: events ncsi_dev_work
[<801073cc>] (unwind_backtrace) from [<8010559c>] (show_stack+0x10/0x14)
[<8010559c>] (show_stack) from [<80110f50>] (__warn+0xd8/0xf4)
[<80110f50>] (__warn) from [<80111060>] (warn_slowpath_null+0x1c/0x24)
[<80111060>] (warn_slowpath_null) from [<8044f690>] (ncsi_start_channel_monitor+0x40/0x78)
[<8044f690>] (ncsi_start_channel_monitor) from [<80450314>] (ncsi_configure_channel+0x294/0x2d4)
[<80450314>] (ncsi_configure_channel) from [<80450acc>] (ncsi_dev_work+0x3a8/0x3d8)
[<80450acc>] (ncsi_dev_work) from [<80122ba0>] (process_one_work+0x18c/0x2f0)
[<80122ba0>] (process_one_work) from [<80122fb4>] (worker_thread+0x280/0x3cc)
[<80122fb4>] (worker_thread) from [<80127888>] (kthread+0x10c/0x124)
[<80127888>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
---[ end trace 6ce2e89ac24c35d5 ]---
```

So probably it is likely on the kernel 4.10 would show up as well..

gwshan, 2017-05-16:
The state tracking if NCSI monitor has been enabled or not, can be messy. There is one patch fixing the issue, but it's not merged to upstream yet. However, the problem is likely caused by far-end network adapter (e.g. BCM5719). That's to say, the far-end NIC doesn't receive the GLS (Get-Link-State) command and send response to BMC. So please check if the host is in good state when seeing the unexpected backtrace on BMC side.

gkeishin, 2017-05-23:
```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 985 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:240 ncsi_start_channel_monitor+0x40/0x78
CPU: 0 PID: 985 Comm: kworker/0:2 Not tainted 4.7.10-15c30d4deb0a01f398230395dc0d9af4be92d856 #1
Hardware name: ASpeed SoC
Workqueue: events ncsi_dev_work
[<8010723c>] (unwind_backtrace) from [<801054c4>] (show_stack+0x10/0x14)
[<801054c4>] (show_stack) from [<80110be0>] (__warn+0xd8/0xf4)
[<80110be0>] (__warn) from [<80110cf0>] (warn_slowpath_null+0x1c/0x24)
[<80110cf0>] (warn_slowpath_null) from [<80436614>] (ncsi_start_channel_monitor+0x40/0x78)
[<80436614>] (ncsi_start_channel_monitor) from [<804372a0>] (ncsi_configure_channel+0x294/0x2d4)
[<804372a0>] (ncsi_configure_channel) from [<80437a68>] (ncsi_dev_work+0x3a8/0x3d8)
[<80437a68>] (ncsi_dev_work) from [<80122008>] (process_one_work+0x18c/0x2f0)
[<80122008>] (process_one_work) from [<8012241c>] (worker_thread+0x280/0x3cc)
[<8012241c>] (worker_thread) from [<801267ec>] (kthread+0xdc/0xf0)
[<801267ec>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
---[ end trace fca5271d1238588f ]---
```
Seeing on Latest master build too.. so looks like its coming back
```
- [x] BUILD ID   : https://openpower.xyz/job/openbmc-build/752/
- [x] VERSION_ID : v1.99.5-66-g47b348b
- [x] Kernel     : Linux witherspoon 4.7.10-15c30d4deb0a01f398230395dc0d9af4be92d856 #1 Tue May 23 11:25:56 UTC 2017 armv6l GNU/Linux
```

gkeishin, 2017-06-02:
another on 
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-31"
VERSION_ID="v1.99.6-31-g792087b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-31"
BUILD_ID="v1.99.6"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Fri Jun 2 04:20:38 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```
```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 3 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:240 ncsi_start_channel_monitor+0x40/0x78
CPU: 0 PID: 3 Comm: kworker/0:0 Not tainted 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1
Hardware name: ASpeed SoC
Workqueue: events ncsi_dev_work
[<801073cc>] (unwind_backtrace) from [<8010559c>] (show_stack+0x10/0x14)
[<8010559c>] (show_stack) from [<80110f50>] (__warn+0xd8/0xf4)
[<80110f50>] (__warn) from [<80111060>] (warn_slowpath_null+0x1c/0x24)
[<80111060>] (warn_slowpath_null) from [<80454f50>] (ncsi_start_channel_monitor+0x40/0x78)
[<80454f50>] (ncsi_start_channel_monitor) from [<80455bd4>] (ncsi_configure_channel+0x294/0x2d4)
[<80455bd4>] (ncsi_configure_channel) from [<8045638c>] (ncsi_dev_work+0x3a8/0x3d8)
[<8045638c>] (ncsi_dev_work) from [<80122ba0>] (process_one_work+0x18c/0x2f0)
[<80122ba0>] (process_one_work) from [<80122fb4>] (worker_thread+0x280/0x3cc)
[<80122fb4>] (worker_thread) from [<80127888>] (kthread+0x10c/0x124)
[<80127888>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
---[ end trace d383b363781446d7 ]---
ftgmac100 1e660000.ethernet eth0: NCSI interface up
root@witherspoon:~#

```

gwshan, 2017-06-05:
As I said again and again, Please check if the host has been put into
*GOOD* state when hitting this issue. It's likely the BMC57xx isn't good
state, meaning the host is in messy state.

gkeishin, 2017-06-05:
@gwshan You mean to see by good host state when this happens.. As far I observed, I dont see host going quisce so I ll take that host in good state.. If there are specific command(s) to execute when this occurs in future or log to log to be collected let me know..

gwshan, 2017-06-05:
@gkeishin Yeah, When hitting this, please check if the host has been up successfully. You should see the regular output from IPMI console at least.

The problem is likely the host isn't up properly and the broadcom chip has been put into unstable state. The chip keeps sending flow control request to BCM in flooding mode, eventually the Rx Ring is exhausted. No more NCSI response, including GLS response used in the channel monitor, can be received successfully.

gkeishin, 2017-06-07:
Here is another on latest build but this case is on boot to standby. 

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-78"
VERSION_ID="v1.99.6-78-gfd4d3cc"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-78"
BUILD_ID="v1.99.6"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Tue Jun 6 22:21:52 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```
- [x] Journald log 
[journald_v1.99.6-78-gfd4d3cc_ncsi_log.txt](https://github.com/openbmc/openbmc/files/1057733/journald_v1.99.6-78-gfd4d3cc_ncsi_log.txt)

```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1002 at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:240 ncsi_start_channel_monitor+0x40/0x78
CPU: 0 PID: 1002 Comm: kworker/0:5 Not tainted 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1
Hardware name: ASpeed SoC
Workqueue: events ncsi_dev_work
[<801073cc>] (unwind_backtrace) from [<8010559c>] (show_stack+0x10/0x14)
[<8010559c>] (show_stack) from [<80110f50>] (__warn+0xd8/0xf4)
[<80110f50>] (__warn) from [<80111060>] (warn_slowpath_null+0x1c/0x24)
[<80111060>] (warn_slowpath_null) from [<80454f50>] (ncsi_start_channel_monitor+0x40/0x78)
[<80454f50>] (ncsi_start_channel_monitor) from [<80455bd4>] (ncsi_configure_channel+0x294/0x2d4)
[<80455bd4>] (ncsi_configure_channel) from [<8045638c>] (ncsi_dev_work+0x3a8/0x3d8)
[<8045638c>] (ncsi_dev_work) from [<80122ba0>] (process_one_work+0x18c/0x2f0)
[<80122ba0>] (process_one_work) from [<80122fb4>] (worker_thread+0x280/0x3cc)
[<80122fb4>] (worker_thread) from [<80127888>] (kthread+0x10c/0x124)
[<80127888>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
---[ end trace a23d80e0d87f3404 ]---
ftgmac100 1e660000.ethernet eth0: NCSI interface up
Apr 20 17:34:37 witherspoon kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 20 17:34:37 witherspoon kernel: ------------[ cut here ]------------
Apr 20 17:34:37 witherspoon kernel: WARNING: CPU: 0 PID: 1002 at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:240 ncsi_start_channel_monitor+0x40/0x78
Apr 20 17:34:37 witherspoon kernel: CPU: 0 PID: 1002 Comm: kworker/0:5 Not tainted 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1
Apr 20 17:34:37 witherspoon kernel: Hardware name: ASpeed SoC
Apr 20 17:34:37 witherspoon kernel: Workqueue: events ncsi_dev_work
Apr 20 17:34:37 witherspoon kernel: [<801073cc>] (unwind_backtrace) from [<8010559c>] (show_stack+0x10/0x14)
Apr 20 17:34:37 witherspoon kernel: [<8010559c>] (show_stack) from [<80110f50>] (__warn+0xd8/0xf4)
Apr 20 17:34:37 witherspoon kernel: [<80110f50>] (__warn) from [<80111060>] (warn_slowpath_null+0x1c/0x24)
Apr 20 17:34:37 witherspoon kernel: [<80111060>] (warn_slowpath_null) from [<80454f50>] (ncsi_start_channel_monitor+0x40/0x78)
Apr 20 17:34:37 witherspoon kernel: [<80454f50>] (ncsi_start_channel_monitor) from [<80455bd4>] (ncsi_configure_channel+0x294/0x2d4)
Apr 20 17:34:37 witherspoon kernel: [<80455bd4>] (ncsi_configure_channel) from [<8045638c>] (ncsi_dev_work+0x3a8/0x3d8)
Apr 20 17:34:37 witherspoon kernel: [<8045638c>] (ncsi_dev_work) from [<80122ba0>] (process_one_work+0x18c/0x2f0)
Apr 20 17:34:37 witherspoon kernel: [<80122ba0>] (process_one_work) from [<80122fb4>] (worker_thread+0x280/0x3cc)
Apr 20 17:34:37 witherspoon kernel: [<80122fb4>] (worker_thread) from [<80127888>] (kthread+0x10c/0x124)
Apr 20 17:34:37 witherspoon kernel: [<80127888>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
Apr 20 17:34:37 witherspoon kernel: ---[ end trace a23d80e0d87f3404 ]---
Apr 20 17:34:37 witherspoon kernel[820]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 20 17:34:37 witherspoon kernel[820]: ------------[ cut here ]------------
Apr 20 17:34:37 witherspoon kernel[820]: WARNING: CPU: 0 PID: 1002 at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:240 ncsi_start_channel_monitor+0x40/0x78
Apr 20 17:34:37 witherspoon kernel[820]: CPU: 0 PID: 1002 Comm: kworker/0:5 Not tainted 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1
Apr 20 17:34:37 witherspoon kernel[820]: Hardware name: ASpeed SoC
Apr 20 17:34:37 witherspoon kernel[820]: Workqueue: events ncsi_dev_work
Apr 20 17:34:37 witherspoon kernel[820]: [<801073cc>] (unwind_backtrace) from [<8010559c>] (show_stack+0x10/0x14)
Apr 20 17:34:37 witherspoon kernel[820]: [<8010559c>] (show_stack) from [<80110f50>] (__warn+0xd8/0xf4)
Apr 20 17:34:37 witherspoon kernel[820]: [<80110f50>] (__warn) from [<80111060>] (warn_slowpath_null+0x1c/0x24)
Apr 20 17:34:37 witherspoon kernel[820]: [<80111060>] (warn_slowpath_null) from [<80454f50>] (ncsi_start_channel_monitor+0x40/0x78)
Apr 20 17:34:37 witherspoon kernel[820]: [<80454f50>] (ncsi_start_channel_monitor) from [<80455bd4>] (ncsi_configure_channel+0x294/0x2d4)
Apr 20 17:34:37 witherspoon kernel[820]: [<80455bd4>] (ncsi_configure_channel) from [<8045638c>] (ncsi_dev_work+0x3a8/0x3d8)
Apr 20 17:34:37 witherspoon kernel[820]: [<8045638c>] (ncsi_dev_work) from [<80122ba0>] (process_one_work+0x18c/0x2f0)
Apr 20 17:34:37 witherspoon kernel[820]: [<80122ba0>] (process_one_work) from [<80122fb4>] (worker_thread+0x280/0x3cc)
Apr 20 17:34:37 witherspoon kernel[820]: [<80122fb4>] (worker_thread) from [<80127888>] (kthread+0x10c/0x124)
Apr 20 17:34:37 witherspoon kernel[820]: [<80127888>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
Apr 20 17:34:37 witherspoon kernel[820]: ---[ end trace a23d80e0d87f3404 ]---

```

```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~#

```

gkeishin, 2017-06-19:
```
Apr 20 17:34:33 witherspoon kernel: ------------[ cut here ]------------
Apr 20 17:34:33 witherspoon kernel: WARNING: CPU: 0 PID: 175 at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:240 ncsi_start_channel_monitor+0x7c/0x8c
Apr 20 17:34:33 witherspoon kernel: CPU: 0 PID: 175 Comm: kworker/0:1 Not tainted 4.10.17-09041b7c3b65a7ec1063d4882b58fd6f00fc87b3 #1
Apr 20 17:34:33 witherspoon kernel: Hardware name: ASpeed SoC
Apr 20 17:34:33 witherspoon kernel: Workqueue: events ncsi_dev_work
Apr 20 17:34:33 witherspoon kernel: [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
Apr 20 17:34:33 witherspoon kernel: [<8000d934>] (show_stack) from [<801dbecc>] (dump_stack+0x20/0x28)
Apr 20 17:34:33 witherspoon kernel: [<801dbecc>] (dump_stack) from [<80018d54>] (__warn+0xe0/0x108)
Apr 20 17:34:33 witherspoon kernel: [<80018d54>] (__warn) from [<80018e48>] (warn_slowpath_null+0x30/0x38)
Apr 20 17:34:33 witherspoon kernel: [<80018e48>] (warn_slowpath_null) from [<803efaa8>] (ncsi_start_channel_monitor+0x7c/0x8c)
Apr 20 17:34:33 witherspoon kernel: [<803efaa8>] (ncsi_start_channel_monitor) from [<803f0644>] (ncsi_configure_channel+0xb8/0x310)
Apr 20 17:34:33 witherspoon kernel: [<803f0644>] (ncsi_configure_channel) from [<803f0e18>] (ncsi_dev_work+0xac/0x494)
Apr 20 17:34:33 witherspoon kernel: [<803f0e18>] (ncsi_dev_work) from [<8002d21c>] (process_one_work+0x1e0/0x450)
Apr 20 17:34:33 witherspoon kernel: [<8002d21c>] (process_one_work) from [<8002d4e8>] (worker_thread+0x5c/0x570)
Apr 20 17:34:33 witherspoon kernel: [<8002d4e8>] (worker_thread) from [<800335ec>] (kthread+0x124/0x164)
Apr 20 17:34:33 witherspoon kernel: [<800335ec>] (kthread) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
Apr 20 17:34:33 witherspoon kernel: ---[ end trace f32cfb4bfbd1da04 ]---
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] ------------[ cut here ]------------
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] WARNING: CPU: 0 PID: 175 at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/net/ncsi/ncsi-manage.c:240 ncsi_start_channel_monitor+0x7c/0x8c
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] CPU: 0 PID: 175 Comm: kworker/0:1 Not tainted 4.10.17-09041b7c3b65a7ec1063d4882b58fd6f00fc87b3 #1
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] Hardware name: ASpeed SoC
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] Workqueue: events ncsi_dev_work
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<8000d934>] (show_stack) from [<801dbecc>] (dump_stack+0x20/0x28)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<801dbecc>] (dump_stack) from [<80018d54>] (__warn+0xe0/0x108)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<80018d54>] (__warn) from [<80018e48>] (warn_slowpath_null+0x30/0x38)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<80018e48>] (warn_slowpath_null) from [<803efaa8>] (ncsi_start_channel_monitor+0x7c/0x8c)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<803efaa8>] (ncsi_start_channel_monitor) from [<803f0644>] (ncsi_configure_channel+0xb8/0x310)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<803f0644>] (ncsi_configure_channel) from [<803f0e18>] (ncsi_dev_work+0xac/0x494)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<803f0e18>] (ncsi_dev_work) from [<8002d21c>] (process_one_work+0x1e0/0x450)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<8002d21c>] (process_one_work) from [<8002d4e8>] (worker_thread+0x5c/0x570)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<8002d4e8>] (worker_thread) from [<800335ec>] (kthread+0x124/0x164)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] [<800335ec>] (kthread) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
Apr 20 17:34:33 witherspoon kernel[908]: [  323.840000] ---[ end trace f32cfb4bfbd1da04 ]---

```

here another one BMC reset to standby
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-12"
VERSION_ID="v1.99.7-12-gc533d5c"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-12"
BUILD_ID="v1.99.7"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-09041b7c3b65a7ec1063d4882b58fd6f00fc87b3 #1 Mon Jun 19 14:21:14 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

```

gkeishin, 2017-06-28:
Waiting for https://patchwork.ozlabs.org/patch/750225/  for this fix

gkeishin, 2017-09-19:
Good to close @shenki @gwshan  this one

closed 2017-09-20 by shenki