openbmc/openbmc #1599: OpenFSI performance improvements

cbostic, 2017-05-11:
Due to occasional FSI protocol stability issues when running hostboot, larger delays were added between clock and data transitions.  Total delay now per FSI clock cycle is 6 microseconds.   Need to understand root cause of this issue and remove as much delay as possible to improve performance.

I've been talking to David Green in Card I/O about probing up these I/O lines to see what may be up.
Tuesday he was able to provide some scope dumps for an FSI transaction.  Nothing unusual was seen from a rise/fall time perspective and no noise was noted.   The timing between when data changes state and when the clock transitions from 0 to 1 is fairly long though and could potentially result in issues when data direction changes to slave -> master mode.

[image]

cbostic, 2017-05-12:
[image]
[image]

cbostic, 2017-05-15:
Accidentally closed this out.  Needs to remain open.

cbostic, 2017-05-25:
I've received a large csv formatted data dump of the waveforms which can be shared.  Too large to attach here.

cbostic, 2017-10-16:
Marking as 'Phase 7'

amboar, 2018-01-08:
> Due to occasional FSI protocol stability issues when running hostboot
This is the limit of the justification for the artificial delays. I'd like to see a greater exposition of the issue(s) to argue against why we can't just remove them.

closed 2019-02-26 by rfrandse