openbmc/openbmc #1600: Sprint 17.05.17 - Committed

rfrandse, 2017-05-11:


bradbishop, 2017-05-15:
-------------------------------------------------------------------------------------------------
***Squad 2 -***
-------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------
**Matt:spinler -** 
**17.01.11:**
- #629 (integrate device tree into kernel) - waiting

**17.04.19:**
- #1346 - tach feedback func state monitoring

**17.05.03:**
  - #1567 - pass arguments to phosphor fan control

-------------------------------------------------------------------------------------------------
**Matt B.:msbarth -**
**17.04.19:**
- #944 (degraded mode)

-------------------------------------------------------------------------------------------------
**Brad:bradbishop -**

**17.03.22:**
- #1303 (inventory Monitoring - Determine design and yaml input layout) 

-------------------------------------------------------------------------------------------------
**Chris -**
**17.03.08:**
- #1219 (OpenFSI targeted error handling for mulitple hub links)
   - 4/24 - in 4.10 submit

**17.03.22:**
- #1222 ( OpenFSI: interrupt support for hub links.)
   - 4/24 - up for review

**17.04.19**
 - need a story
 - 4/25 working on an FSP defect
 - #1536 (document raw/scan files)
    - 05/02 - mostly done
    - 05/04 - v7 ready to go

- interrupt support for 4.10.  need a story

- 05/11 - made some upstream feedback changes

**17.05.03**
  - #1599 (perf improvements)
      - 05/12 - started working this

------------------------------------------------------------------------------------------------
**Eddie:eddiejames -**
**17.03.22:**
- #1330 (OCC: DT bindings for occ-hwmon) 

**17.04.19:**
- #1492 (4.10 kernel integration)
- #1526 - pdbg updates
-------------------------------------------------------------------------------------------------
**Brandon Wyman -**
**17.03.22:**
- #370 ( Fan control - Air cool vs Water cool policies.)

-------------------------------------------------------------------------------------------------
**Dinesh Chinari  -**
- #1315 (Ensure uboot fw_env update script uses new inventory object) 

- #1322 (pgood is lost on BMC reboot on witherspoon)

dkodihal, 2017-05-15:
Squad 1 sprint plan

**Deepak**
**17.05.03**
#1557  Design proposal to preserve and restore d-bus objects

**Ratan**
**17.05.17**
#1478  mboxd: Handle multiple mount types

**Devender**
**17.05.17**
#1560  IPMI FRU commands : implementation

**Vishwa**
**17.05.17**
#1363 Request OCC reset when fail detected

**Jayanth**
**17.05.17**
#1504 BMC Dump: Create inotify daemon
#1510 BMC Dump: Add support for User initiated Dump

anoo1, 2017-05-15:
Squad 3 sprint plan

**Adriana**
**17.05.03**
#1297 eSEL Missing AdditionalData [ Witherspoon ]
#? Systemd service file to delete volumes
#? Obmc image packaging

**Gunnar**
**17.05.17**
#?

**Andrew**
**17.05.17**
#?

**Michael**
**17.05.17**
#865 Factory reset

**Saqib**
**17.05.17**
#? End-to-end code update verification

**Leo**
**17.05.17**
#?

closed 2017-06-08 by rfrandse