openbmc/openbmc #1602: "Change Root test failed!" after reset of witherspoon

geissonator, 2017-05-13:
Was doing a normal code update and had the serial console up.  Not sure what code level was on the system prior but this is what I get in serial, even after a PDU cycle.

Probably not much we can do other then reflash the chip manually but figured I'd get a bug open to track this in case it pops up elsewhere.  System was w42.

Continuing normal startup doesn't work and I didn't see anything obvious in maintenance mode.

```
U-Boot 2016.07 (Mar 29 2017 - 18:57:19 +0000)

DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Error: aspeednic#0 address not set.

Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x2008012c
     Data Size:    2016552 Bytes = 1.9 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha1
     Hash value:   5b07aa90b1754af746281f26ae4d9fe0dc789463
   Verifying Hash Integrity ... sha1+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'ramdisk@1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  lzma compressed
     Data Start:   0x2027267c
     Data Size:    1586388 Bytes = 1.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha1
     Hash value:   2ce649591ce56f00379fbca45d29d18c76787919
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'fdt@1' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x2026c748
     Data Size:    24180 Bytes = 23.6 KiB
     Architecture: ARM
     Hash algo:    sha1
     Hash value:   7f630ccf0371f494e9b619cfda760d0b73ffb9a3
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x2026c748
   Loading Kernel Image ... OK
   Loading Ramdisk to 9ea05000, end 9eb884d4 ... OK
   Loading Device Tree to 9e9fc000, end 9ea04e73 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 (jenkins@openbmc-builder1) (gcc version 6.2.0 (GCC) ) #1 Fri May 12 15:25:15 UTC 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[... 119 lines cut ...]
[    4.330000] rtc-rv8803 11-0032: setting system clock to 2016-03-01 13:12:14 UTC (1456837934)
[    4.330000] Freeing unused kernel memory: 272K
[    4.330000] This architecture does not have kernel memory protection.
rofs = mtd4 squashfs rwfs = mtd5 jffs2
copy-files-to-ram copy-base-filesystem-to-ram
update: --copy-files --clean-saved-files --no-restore-files
[    6.460000] jffs2: notice: (467) jffs2_build_xattr_subsystem: complete building xattr subsystem, 1 of xdatum (1 unchecked, 0 orphan) and 19 of xref (0 dead, 7 orphan) found.
Skipping empty update of rwfs.
copy-files-to-ram copy-base-filesystem-to-ram
mount: can't setup loop device: No such device or address
Running with read-write overlay in RAM for this boot.
No state will be preserved unless flash update performed.
chroot: can't execute '/bin/sh': No such file or directory

Unable to confirm /sbin/init is an executable non-empty file
in merged file system mounted at /root.

Change Root test failed!  Invoking emergency shell.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup):

```

williamspatrick, 2017-05-13:
Looks like the RO-fs became corrupted.  Couldn't find /bin/sh or /sbin/init.

geissonator, 2017-05-15:
Another system, w50, hit this issue as well.  Maybe something to do with moving between the 4.7 and 4.10 kernels?

geissonator, 2017-05-15:
2 more systems hit this today.  The issue is related to the 4.10 changes (https://gerrit.openbmc-project.xyz/#/c/2846/).  It flashes ok onto the system, and tests work fine.  "reboot" works fine as well.  But as soon as we do a prepareForUpdate REST call (which reboots the BMC into a ramdisk env), we hit the above fail and do something there that breaks the flash to the point where power cycles do not fix things.

mdmillerii, 2017-05-16:
So we find the mtd partitions for rofs and rwfs:
'''
rofs = mtd4 squashfs rwfs = mtd5 jffs2
'''
We seem to be able to mount the read/write jffs2 without complaints:
'''
update: --copy-files --clean-saved-files --no-restore-files
[    6.460000] jffs2: notice: (467) jffs2_build_xattr_subsystem: complete building xattr subsystem, 1 of xdatum (1 unchecked, 0 orphan) and 19 of xref (0 dead, 7 orphan) found.
Skipping empty update of rwfs.
'''

But get -ENODEV when we try to mount /dev/mtdblock4:
'''
mount: can't setup loop device: No such device or address
'''
'''
Change Root test failed!  Invoking emergency shell.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup)
'''
and ask for the root password on the serial port to debug.

I would check the kernel config for squashfs, xz, and mtdblock to start.  Otherwise login with the standard password and look around.

mdmillerii, 2017-05-16:
perhaps the uboot env is erasing too large a chunk and killing the beginning of the squashfs?  We turned off 4k spi-nor for ubi, and the env is 6000-8000.

If so we can tweak the fwenv to rmw the env block for the short term... we really want 2 64k sectors though.

mdmillerii, 2017-05-16:
thinking more , its likely the loop block device is missing.

shenki, 2017-05-17:
Fixed in the latest version of the move-to-4.10 patch: https://gerrit.openbmc-project.xyz/#/c/2846/

eddiejames, 2017-05-17:
i just hit this on barreleye, which already had the loop device enabled.

cbostic, 2017-05-19:
mdmillerii  mentioned some possibilities on -ENOENT issue:

"The obvious place -ENOENT is the table of spi device identifiers and jedec id's.  The next step would be to capture the dmesg from a working kernel to find the flash that is identified and make sure its in the list in the 4.10 branch."
 

dmesg from a Witherspoon 4.10 boot:

```console [ttyS4] enabled
1e787000.vuart: ttyS2 at MMIO 0x1e787000 (irq = 22, base_baud = 1500000) is a 16550A
timeriomem_rng 1e6e2078.hwrng: 32bits from 0xf00e2078 @ 1us
brd: module loaded
loop: module loaded
aspeed-lpc-ctrl 1e789080.lpc-ctrl: Loaded at 0x98000000 (0x04000000)
aspeed-smc 1e620000.flash-controller: Using IRQ 18
aspeed-smc 1e620000.flash-controller: DMA support disabled.
aspeed-smc 1e620000.flash-controller: mx25l25635e (32768 Kbytes)
aspeed-smc 1e620000.flash-controller: CE0 window [ 0x20000000- 0x22000000 ] 32MB
aspeed-smc 1e620000.flash-controller: CE1 window [ 0x22000000- 0x2a000000 ] 128MB
5 ofpart partitions found on MTD device bmc
Creating 5 MTD partitions on "bmc":
0x000000000000-0x000000060000 : "u-boot"
0x000000060000-0x000000080000 : "u-boot-env"
0x000000080000-0x0000004c0000 : "kernel"
0x0000004c0000-0x000001c00000 : "rofs"
0x000001c00000-0x000002000000 : "rwfs"
aspeed-smc 1e620000.flash-controller: mx25l25635e (32768 Kbytes)
aspeed-smc 1e620000.flash-controller: CE1 window [ 0x22000000- 0x24000000 ] 32MB
aspeed-smc 1e620000.flash-controller: CE2 window [ 0x24000000- 0x2c000000 ] 128MB
aspeed-smc 1e630000.flash-controller: mx66l1g45g (131072 Kbytes)
aspeed-smc 1e630000.flash-controller: CE0 window resized to 120MB (AST2500 HW quirk)
aspeed-smc 1e630000.flash-controller: CE0 window [ 0x30000000- 0x37800000 ] 120MB
aspeed-smc 1e630000.flash-controller: CE0 window too small for chip 128MB
aspeed-smc 1e630000.flash-controller: CE1 window [ 0x37800000- 0x38000000 ] 8MB
libphy: Fixed MDIO Bus: probed
```

mdmillerii, 2017-05-20:
The log from @eddiejames had: 
```
aspeed-smc: probe of 1e620000.fmc failed with error -2
aspeed-smc: probe of 1e630000.spi failed with error -2
```
We need to identify the expected modules in the failing system.  That can be accomplished by booting a working (presumably 4.7) kernel.  Alternatively we can instrument [spi_nor_scan()](https://github.com/openbmc/linux/blob/dev-4.10/drivers/mtd/spi-nor/spi-nor.c#L1331) with `hex_dump_to_buffer()` of the id string, at least if it doesn't match.

eddiejames, 2017-05-23:
For Barreleye it was a problem of missing clock phandles in the devicetree for the pnor nodes.

shenki, 2017-06-26:
I believe this is fixed. Please re-open if this is still an issue.

closed 2017-06-26 by shenki