openbmc/openbmc #1604: phosphor-fan-monitor@0.service: Main process exited, code=dumped, status=11/SEGV

geissonator, 2017-05-13:
Something still not quite right with the latest fan monitor service.

We get this seg fault during boot.  And keep restarting the service and getting it again.

Code level is VERSION_ID="v1.99.5-40-gbaf766c"

System seems to eventually boot ok

```
Apr 22 09:25:41 witherspoon systemd[1]: Started Scan FSI devices.
Apr 22 09:25:41 witherspoon systemd[1]: Starting Run VCS workaround on host0...
Apr 22 09:25:41 witherspoon systemd[1]: phosphor-fan-monitor@0.service: Main process exited, code=dumped, status=11/SEGV
Apr 22 09:25:41 witherspoon systemd[1]: phosphor-fan-monitor@0.service: Unit entered failed state.
Apr 22 09:25:41 witherspoon systemd[1]: phosphor-fan-monitor@0.service: Failed with result 'core-dump'.
Apr 22 09:25:41 witherspoon systemd[1]: phosphor-fan-monitor@0.service: Service hold-off time over, scheduling restart.
Apr 22 09:25:41 witherspoon systemd[1]: Stopped Phosphor Fan Monitor Daemon.
Apr 22 09:25:41 witherspoon systemd[1]: Started Phosphor Fan Monitor Daemon.
Apr 22 09:25:41 witherspoon openpower-proc-control[1218]: Running P9 procedure vcsWorkaround
Apr 22 09:25:41 witherspoon systemd[1]: Started Run VCS workaround on host0.
Apr 22 09:25:41 witherspoon systemd[1]: Starting Turn VCS on...
Apr 22 09:25:42 witherspoon systemd[1]: Started Turn VCS on.
Apr 22 09:25:42 witherspoon systemd[1]: Reached target Chassis0 (Power On).
Apr 22 09:25:42 witherspoon systemd[1]: Reached target Start Host0 (Pre).
Apr 22 09:25:42 witherspoon systemd[1]: Starting Start host0 SBE...
Apr 22 09:25:42 witherspoon phosphor-chassis-state-manager[916]: Received signal that power ON is complete
Apr 22 09:25:42 witherspoon phosphor-chassis-state-manager[916]: Change to Chassis Power State
Apr 22 09:25:42 witherspoon openpower-proc-control[1234]: Running P9 procedure startHost
```

spinler, 2017-05-14:
phosphor-fan.inc needs https://github.com/openbmc/phosphor-fan-presence/commit/84201119da1a5df30b9155de1d9c302d0ec021dd

This bump commit will do it: https://gerrit.openbmc-project.xyz/#/c/4194/

williamspatrick, 2017-05-15:
Resolved with bdd2b66d5070e53c7043328432a2032e6b299c70

closed 2017-05-15 by williamspatrick