openbmc/openbmc #1609: Non-sensor refactor : switch to new interfaces

dkodihal, 2017-05-15:
Switch existing code to new software sensor d-bus interfaces.

geissonator, 2017-07-14:
Dhruv, can you confirm this is a dup of your other work in this area?

dhruvibm, 2017-07-14:
All changes for non-sensors will be implemented with 1608, but is there any cleanup needed for old interfaces?

geissonator, 2017-07-14:
I'll retitle this to removing the old code and taking out of beta/phase 5.

bradbishop, 2017-09-25:
https://gerrit.openbmc-project.xyz/#/c/6782/1/sensorhandler.cpp:

```Tom Joseph
Sep 20 9:28 AM
Please add a note in #1609, in an earlier review i had commented that #2154 is last legacy sensor and the case statement can be removed after that.```

@tomjoseph83 @dhruvibm ^

anoo1, 2018-09-12:
Removing the legacy functions with https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-host-ipmid/+/12759/

geissonator, 2019-02-25:
Looks done to me

closed 2019-02-25 by geissonator