openbmc/openbmc #1613: Add DTS310 instance to Witherspoon device tree

bradbishop, 2017-05-15:
Not sure if the driver is in the tree yet or not.  This story may be blocked on that happening first.

bjwyman, 2017-05-16:
Does the title of this story have a typo?
Are you referring to the DPS310 ambient temperature sensor?

cbostic, 2017-05-18:
- [ ] Locate device drivers available for DTS310
- [ ] Obtain card schematics including details on DPS310 hookup.
- [ ] Identify i2c bus and address details for Witherspoon
- [ ] Code and integrate changes to dts file.

cbostic, 2017-05-18:
Looks like 4.10 kernel already has dps310 listed in the dev tree.   Address looks incorrect however based on feedback from Gene Davison.

```       dps310@76 {
                compatible = "infineon,dps310";
                reg = <0x76>;
        };
```
Should be 0xEC

mdmillerii, 2017-05-18:
Looking at the bmp280 , the i2c address should be the 7-bit right aligned value, not the 8-bit with r/w bit in the bottom, so 0x76 would be correct.  Did you try to check this with i2c-tool or check for kernel probe finding the device?

cbostic, 2017-05-18:
>
> Looking at the bmp280 , the i2c address should be the 7-bit right 
> aligned value, not the 8-bit with r/w bit in the bottom, so 0x76 would 
> be correct. Did you try to check this with i2c-tool or check for 
> kernel probe finding the device?
Hi Milton,

Thanks for the clarification.  I wasn't aware it was in 7-bit right 
aligned format.  dps310 driver doesn't exist as far as I can see so 
wouldn't have probed.  Could have checked with i2c-tool though.

Cancelling this.

Thanks
Chris

cbostic, 2017-05-18:
Appears this was updated in 4.10 kernel last month by @shenki     Were there other updates required here?
```
commit f11965f6f5ee6d6f2f9f46b7e9db90964c0f046c
Author: Joel Stanley <joel@jms.id.au>
Date:   Tue Apr 18 14:11:55 2017 +0930

    ARM: dts: witherspoon: Add Infineon DSP310 i2c part
    
    Signed-off-by: Joel Stanley <joel@jms.id.au>

diff --git a/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts b/arch/arm/boot/dts/aspeed-bmc-opp
index b9a337e..7ef7b0d 100644
--- a/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts
+++ b/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts
@@ -172,6 +172,11 @@
                compatible = "maxim,max31785";
                reg = <0x52>;
        };
+
+       dps310@76 {
+               compatible = "infineon,dps310";
+               reg = <0x76>;
+       };
 };
```

bradbishop, 2017-05-19:
Just FYI driver work being tracked here: #1196

bradbishop, 2017-05-19:
@cbostic looks like we are good here and can just track the overall move to 4.10.

bradbishop, 2017-05-19:
Already done, closing...

closed 2017-05-19 by bradbishop