openbmc/openbmc #1616: Add onfailure to ubi systemd service files

anoo1, 2017-05-15:


anoo1, 2017-05-18:
- [ ] Call the ubi unmount service if the ubi mount service fails.
- [ ] Create a squashfs unmount service to do the opposite of the squashfs mount service: unmount the ro mount and remove the volume.
- [ ] Call the squashfs unmount service if the mount service fails.

saqibkh, 2017-09-06:
PNOR:
- [ ]  Combine the services that creates pnor-ro<versionId> and pnor-rw<versionId>
- [ ]  Add OnFailure to the obmc-flash-bios-ubimount@.service to call ubi delete services (umount ubi volumes, delete mountpoint under /media and then delete the ubi volumes)

BMC:
- [ ]  Add OnFailure=obmc-flash-bmc-ubiro-remove@%i.service to the obmc-flash-bmc-ubiro@.service.

rfrandse, 2017-09-07:
<https://gerrit.openbmc-project.xyz/6348> Handle the case where obmc-flash-bmc-ubimount@.service fails
Resolves  #1616 Add onfailure to ubi systemd service files

closed 2017-09-12 by williamspatrick