openbmc/openbmc #1617: phosphor-logging: Cap the number of error logs

anoo1, 2017-05-15:
The old records.events implementation capped the number of error logs to 128 via a systemd service file configuration setting. If the limit is reached, no more error logs are created.

For the phosphor-logging implementation, probably need to do a mapper query to get the number of error logs, then determine if the log is allowed.

References:

https://github.com/openbmc/openbmc/blob/master/meta-phosphor/common/recipes-phosphor/logging/obmc-phosphor-event/eventd.conf

https://github.com/openbmc/phosphor-event

sivassrr, 2017-05-16:
May be we should give a warning / informative error log around when we reach 115 or 120 number of error logs.  Then user get an opportunity to clear and create space for new error logs. @anoo1 Hope it is make sense.

lgon, 2017-05-19:
Tasklist
- [ ] - Write a method to get error log count that is called when an error log is created.
- [ ] - Send warning when reach 115 error logs.
- [ ] - Clear oldest log when limit has been reached.

anoo1, 2017-05-31:
Per Patrick, systemd has a way to set a limit based on filesystem percentage.

devenrao, 2017-08-01:
Copied tasks from IGON
 
- [ ] Write a method to get error log count that is called when an error log is created.
- [ ]  Send warning when reach 115 error logs.
- [ ] Clear oldest log when limit has been reached.
Design discussion
- [ ] Need to discuss particks proposal "systemd has a way to set a limit based on filesystem percentage."
# **STATUS**
27/07 to 01/08
On vacation
02/08
Started analyzing the story

rfrandse, 2017-08-10:
<https://gerrit.openbmc-project.xyz/5808> Implement a cap on the number of committed errors
Resolves  #1617 phosphor-logging: Cap the number of error logs

closed 2017-08-16 by williamspatrick