openbmc/openbmc #1619: OCC HWMON support for non-hwmon sysfs attributes

bradbishop, 2017-05-16:
Need to expose:

dvfs due to overtemp
dvfs due to power
mem throttle overtemp
n power

booleans from the OCC as non-hwmon sysfs attributes.

eddiejames, 2017-05-18:
- [ ] export sysfs attributes from platform driver device, not hwmon device
- [ ] test it

geissonator, 2017-06-01:
Create follow on story to use non-sensor sensors implementation to expose these?

eddiejames, 2017-06-05:
There is no "n power" (which is redundant power supply present) sensor on P9 occ. We have "quick power drop" instead.

geissonator, 2017-06-15:
This includes providing hwmon entry on which occ is the master

geissonator, 2017-06-20:
Opened #1821 to track the use of these new hwmon sensors.

rfrandse, 2017-07-06:
This is done. Eddie sent patch via mailing list  for openbmc linux

closed 2017-07-06 by rfrandse