openbmc/openbmc #1620: HOST reaches quiesce state after host power on- Bareleye

prkatti1, 2017-05-16:
bareleye system reaches quiesce state when we power on.

System info
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-41"
VERSION_ID="v1.99.5-41-gbdd2b66"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-41"
BUILD_ID="v1.99.5"

root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced

[xcat_info.txt](https://github.com/openbmc/openbmc/files/1004073/xcat_info.txt)
[qemu_creation.txt](https://github.com/openbmc/openbmc/files/1004074/qemu_creation.txt)

sivassrr, 2017-05-16:
@prkatti1 pls capture FFDC and also update PNOR details.

geissonator, 2017-05-16:
Yes, we need the journal.  Please reopen once available.  Also, 9 times out of 10, quiesce is related to a bad PNOR image, please be sure you have the latest tested one on there.

closed 2017-05-16 by geissonator