openbmc/openbmc #1621: phosphor-fan-monitor: Setting a fan to nonfunctional [Witherspoon]

gkeishin, 2017-05-17:
This is specific for BMC Kernel 4.10

**BMC Driver :**

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-45"
VERSION_ID="v1.99.5-45-g386583b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-45"
BUILD_ID="v1.99.5"
root@witherspoon:~#
```

**PNOR:**
```
IBM-witherspoon-ibm-OP9_v1.16_1.3
        op-build-v1.16-35-g2bb0a83-dirty
        buildroot-2017.02-4-gc28573e
        skiboot-5.5.0
        hostboot-c9bd42f
        linux-4.10.7-openpower1-p61306aa
        petitboot-v1.4.2-p58d71a5
        witherspoon-xml-3a7d05c
        occ-e1889e5
        hostboot-binaries-8cbbc67
        capp-ucode-1bb7503
        sbe-9b98de3-p554c410
```

```
Apr 20 17:31:45 witherspoon phosphor-fan-monitor[1106]: Setting a fan to nonfunctional
Apr 20 17:31:45 witherspoon phosphor-fan-monitor[1106]: Setting a fan to nonfunctional
Apr 20 17:31:46 witherspoon phosphor-fan-monitor[1106]: Setting a fan to nonfunctional
Apr 20 17:31:46 witherspoon phosphor-fan-monitor[1106]: Setting a fan to nonfunctional
```

- [x] Journald 
[journald_v1.99.5-45-g386583b_fan_disabled.txt](https://github.com/openbmc/openbmc/files/1007778/journald_v1.99.5-45-g386583b_fan_disabled.txt)

@eddiejames @geissonator

spinler, 2017-05-17:
This is by design.  This first story that went in just puts in fan-monitor and starts it at power on, where the app gives fans 15 seconds to get to full speed (not enough time).  With issue https://github.com/openbmc/openbmc/issues/1567, I am first giving the fans 20s on a power on before the monitor gives it another 15.  This isssue is in review now.

We have code in the pipeline that will actually look at this property and respond accordingly, but it isn't there yet which is why I integrated the current code knowing this would happen.

spinler, 2017-05-17:
Closing this as working as designed.  When issue 1567 goes in then if we see it we may need to adjust some timeout numbers.

closed 2017-05-17 by spinler