openbmc/openbmc #1626: Fan Control: Calculate speed change request

msbarth, 2017-05-17:
Each sensor involved with fan target speed changes, a T-control(temperature) high and T-control(temperature) low value needs to be defined to determine the net speed change request for each sensor by some increase RPM delta and decrease RPM delta. A live speed increase request value should be stored on each zone containing the largest net speed change based on the temperature sensor readings.

- [ ] Create a "sensors" yaml file containing the TctrlHigh, TctrlLow, increase delta(per degree), and decrease delta(per degree) for a list of sensors and the zone(s) they are associated with
- [ ] Parse "sensors" yaml file into a "SensorDefinition" for each zone
- [ ] Also parse the "sensors" yaml file into the events so each sensor's property change signal is subscribed to and updated in the zone's property list
- [ ] Add saving last known target speed set(handle application restarts, i.e. fullspeed)
- [ ] Create a function within the zone that calculates the new target speed based on the property value stored for each sensor and the defined Tctrl(High/Low) / RPM delta.

bradbishop, 2017-06-27:
This was not meant to be in 6.14 committed.  I'm not sure why it got added.  Removed.

rfrandse, 2017-06-29:
<https://gerrit.openbmc-project.xyz/4925> Support speed decrease events
Resolves  #1626 Fan Control: Calculate speed change request

closed 2017-07-07 by williamspatrick