openbmc/openbmc #1627: Save the systemd journal to a persistent storage

anoo1, 2017-05-17:
Currently the system journal is stored in memory so it gets deleted when the BMC is rebooted.
Need to change its settings so that it gets stored in a persistent storage in the BMC. Also a size restriction and policy upon reaching that limit would need to be set.

geissonator, 2017-07-19:
I think at least the size restriction aspect of this story needs to be done in beta.

geissonator, 2017-07-24:
Per design meeting, we need to determine if we have the space to write to persistent storage and what the current settings are for max space usage by the journal.

spinler, 2017-07-26:
The documentation says the default is to use up to 10%, while keeping free at least 15%.  This is true for both volatile and persistent settings.  Is that good enough?

spinler, 2017-07-26:
Also, I tried setting Storage=persistent in journald.conf.  It then saves to `/var/log/journal`.  However, that doesn't actually persist either.
```
# journalctl --list-boots --no-pager
 0 84270fd71a6a4cbe8a92beccb0a1e680 Wed 2017-07-26 20:36:28 UTC—Wed 2017-07-26 20:37:46 UTC
```

spinler, 2017-07-26:
# **STATUS**
* 7/26:  Tried persisting journal.  It didn't persist.  Want to talk in the design meeting about if this is actually what we want or not.

mdmillerii, 2017-07-26:
A yocto recipe symlinks `/var/log` to `/var/volatile/log` and starts the journal after mounting `tmpfs` on `/var/volatile`.

spinler, 2017-07-27:
Design meeting discussion:  Decided not to persist.  Instead, will investigate if the default of 10% max usage of filesystem will affect performance or not.

spinler, 2017-07-27:
Test results, from w15:

```
# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-98"
VERSION_ID="v1.99.8-98-g06734f3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-98"
BUILD_ID="v1.99.8"
```

```
# journalctl --disk-usage
Archived and active journals take up 8.0M in the file system.
```

At standby:
```
# free
             total       used       free     shared    buffers     cached
Mem:        432276     137296     294980      14448      10240      48636
-/+ buffers/cache:      78420     353856
Swap:            0          0          0
```

At host login prompt:
```
# free
             total       used       free     shared    buffers     cached
Mem:        432276     141524     290752      14636      10424      49544
-/+ buffers/cache:      81556     350720
Swap:            0          0          0
```

rfrandse, 2017-10-11:
<https://gerrit.openbmc-project.xyz/7202> Set maximum journald size to 1MB on witherspoon
Resolves  #1627 Save the systemd journal to a persistent storage

geissonator, 2017-10-12:
Some notes on what I've found so far with my testing.

journald starts off by writing to volatile disk (/run/log/journal/).  This is so even the earliest services can log to the journal.  During the boot, systemd-journal-flush.service gets run which tells journald to switch over to persistent storage if available.

The issue I'm running into is that when this service runs, the bmc has not yet mounted /etc/machine-id.  The machine-id is used to create the subdirectories under /var/log/journal/.  When this file is not available, journald makes one up.  This causes issues because journalctl uses the /etc/machine-id to know which sub-directory to look under for journal data.  When it looks, /etc/machine-id has been mounted and is the correct value but journald is not writing into that location.

This seems specific to our process for mounting things.  My next steps are to see if I can get systemd-journal-flush.service to wait for /etc/machine-id to be mounted.  Maybe a After=local-fs.target?  Alternative ideas appreciated.

mdmillerii, 2017-10-12:
Make the trigger to switch the journal depend on our `/etc` overlay mount unit.

Definitely related to our cheating on not having `/etc/` read-write when `systemd` starts.

@eddiejames

geissonator, 2017-10-13:
Tried a few things but looks like I'm going to need a feature from journald to make this work.  Opened https://github.com/systemd/systemd/issues/7092 with a proposal.

amboar, 2017-10-16:
From the response I don't think systemd/systemd#7092 is ever going to happen.

closed 2017-10-30 by geissonator

vmauery, 2018-08-30:
Are we just throwing out pie-in-the-sky ideas here? I would like to have a journal that doesn't just limit by size, but rather by date and urgency, and finally by size.

1. Throw out debug entries older than a day.
2. Throw out info entries older than three days.
3. Throw out warning entries older than a week.
4. Truncate the log at one month.
5. Check for size. If the journal is too big, cut the date limits in half and go to 1.

This gives you a level-of-detail view of the system. You see debug, info, and warning messages for the near history, while you keep as many error and above as space permits.

mine260309, 2018-08-31:
Optionally, for systems with SDHC (so we can insert Micro-SDCard), we can move the above entries to external storage instead of throwing away.