openbmc/openbmc #1632: [Barreleye *] Need "obmcutil getsensors" or similar from command line of BMC

yadron, 2017-05-18:
I've noted that some days ago "obmcutil getsensors" removed, see
https://github.com/openbmc/skeleton/commit/53d93023fd8782c9618c1e2b5f08430eedc11ee8

Are there any other way for reading sensors using console of OpenBMC?
I could use
`curl -b cjar -k https://172.17.27.46/org/openbmc/sensors/temperature/RTC_temp1`
, but sometimes it is simpler reading sensors locally using console.

I've tried reverting back these lines to obmcutil
(
```
-    'getsensors': {
-        'bus_name': 'org.openbmc.Sensors',
-        'object_name': '/org/openbmc/sensors',
-        'interface_name': 'org.freedesktop.DBus.ObjectManager',
-        'method': 'GetManagedObjects',
-        'transform': merge_interfaces
-    },
```
), but it doesn't work saying

```
root@barreleye:~# obmcutil getsensors
Traceback (most recent call last):
  File "/usr/sbin/obmcutil", line 90, in <module>
    'transform': merge_interfaces
NameError: name 'merge_interfaces' is not defined
```

Any advises?

williamspatrick, 2017-05-18:
This was removed for two reasons:
   1. We had a policy discussed in the mailing list to eliminate CLI wrappers like this:  https://lists.ozlabs.org/pipermail/openbmc/2016-November/005307.html
   2. The /org/openbmc namespace is deprecated and either replaced or being replaced by content in /xyz/openbmc_project.  The previous tool "sensor" concept was heavily based on IPMI and many of the old "sensors" are no longer in the /.../sensor namespace because they aren't really sensors.

The better option in my mind is to use busctl get-property or busctl introspect to read the sensor values you are interested in.

closed 2017-05-18 by williamspatrick

yadron, 2017-05-23:
Ok, this is an example for retrieving sensor value:

```
busctl call org.openbmc.Sensors /org/openbmc/sensors/temperature/cpu0 org.openbmc.SensorValue getValue
v i 52000
```

williamspatrick, 2017-05-23:
This would be the preferred mechanism with the new xyz namespace objects.  You can also get this via REST.

```
root@palmetto:~# busctl get-property xyz.openbmc_project.Hwmon.hwmon0 /xyz/openbmc_project/sensors/temperature/rtc xyz.openbmc_project.Sensor.Value Value
x 23500
```