openbmc/openbmc #1633: Add support for hwmon labels to phosphor-hwmon

bradbishop, 2017-05-18:
At present the hwmon application ignores the itemX_label sysfs attribute.  hwmon names DBus objects based on the index (the 'X' in itemX_label) only.  

This story tracks the effort to add a level of indirection such that the label property is included in the object naming procedure.

bradbishop, 2017-05-18:
possible implementation...

Consider an existing label:
LABEL_temp1=ambient

This would result an ambient object getting its readings from temp1_input.

For a chip using labels the config file could consist of:

MODE_temp1=label
LABEL_temp121=ambient

The assumption in this example is that the content of temp1_label is 121.  hwmon could be updated to first check for a MODE_% key in the config file.  If it is not found, default to the existing behavior.  If it is found, read the label and then check LABEL_% to get the object name.

bradbishop, 2017-05-18:
additional background here: #1183

tomjoseph83, 2017-06-01:
- [ ] Check if the label is present for the sensor , then go ahead and read the value of the environment variable LABEL_% and MODE_% and return the name of the sensor to create the hwmon dbus object
- [ ] Write testcases to test sensors with label values

rfrandse, 2017-06-13:
<https://gerrit.openbmc-project.xyz/4729> Add a level of indirection when naming the DBUS objects
Resolves  #1633 Add support for hwmon labels to phosphor-hwmon

closed 2017-06-14 by williamspatrick