openbmc/openbmc #1635: Expose non-hwmon MAX31785 sysfs attributes

bradbishop, 2017-05-19:
The MAX31785 has a config register setting that tells it to watch an input and when that input changes to run the fans at full speed (this input is wired to the ast2500 watchdog on witherspoon systems).

A non-hwmon sysfs attribute that exposes this is needed.  Please don't forget to add documentation to Documentation/ABI.  I believe this work can occur directly upstream on lkml.

spinler, 2017-05-19:
It's bit 0 in 0xD9 (MFR_FAULT_RESPONSE).

cbostic, 2017-05-22:
Task list:

- [ ] Identify proper 'struct device' to hang the sysfs file on - needs to be an i2c device type

- [ ] Implement the sysfs changes

- [ ] Verify sysfs operations alter the fan behavior as expected

- [ ] Identify how to merge this change into the pending 4.10 max31785.c driver

- [ ] Send code up to lkml for review

cbostic, 2017-05-26:
I've placed this sysfs file in the i2c dev path:

/sys/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a100.i2c-bus/i2c-3/3-0052/fault_resp

If its required to be somewhere else please advise.

amboar, 2017-05-29:
Does this need to change at runtime? Similar to the watchdog configuration, my gut feeling is it should be devicetree-based.

spinler, 2017-05-30:
No, it doesn't need to change at runtime.

spinler, 2017-06-02:
@cbostic: Was thinking...  If we do a reboot at runtime, that should trigger the watchdog.  When the driver reloads, will it set this bit back to a 0 and turn off driving 100% PWM until we set it back again?  Or will it leave it how it was?

cbostic, 2017-06-06:
>
> @cbostic <https://github.com/cbostic>: Was thinking... If we do a 
> reboot at runtime, that should trigger the watchdog. When the driver 
> reloads, will it set this bit back to a 0 and turn off driving 100% 
> PWM until we set it back again? Or will it leave it how it was?
Matt,

Default is off on driver reload.

spinler, 2017-06-06:
@cbostic Hmm, so we might have a bit of a glitch in the fans, in between the driver loading and then attribute getting set.  Hopefully won't be an issue.

cbostic, 2017-07-19:
This can be closed.

closed 2017-07-19 by rfrandse