openbmc/openbmc #1638: PNOR suddenly becomes invalid, power cycle "fixes" it

geissonator, 2017-05-19:
We've seen this issue across multiple systems running the latest kernel 4.7 based openbmc master.  The system suddenly will not boot and access to PNOR starts failing with pflash.  

```
[witherspoon]# flash_eraseall /dev/mtd7
Erasing 4 Kibyte @ 8000000 - 100% complete.
[witherspoon]# 
[witherspoon]# 
[witherspoon]# pflash -E -f -p /tmp/witherspoon.pnor 
About to erase chip !
Erasing... (may take a while !) done !
About to program "/tmp/witherspoon.pnor" at 0x00000000..0x04000000 !
Programming & Verifying...
[==================================================] 100% ETA:0s     
[witherspoon]#  pflash -r /dev/stdout -S 0 -P VERSION  
FFS: Flash header not found. Code: 100
Error 100 opening ffs !

[witherspoon]#  pflash -i                              
Flash info:
-----------
Name          = /dev/mtd7
Total size    = 128MB 
Erase granule = 4KB 

TOC@0x00000000 Partitions:
-----------
FFS: Flash header not found. Code: 100
Error 100 opening ffs !
```

```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-59"
VERSION_ID="v1.99.5-59-g3e1b846"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-59"
```

```
# system in fail state
root@witherspoon:/tmp# hexdump -C /dev/mtd7 |more
00000000  d0 41 52 54 00 00 00 01  00 00 00 01 00 00 00 80  |.ART............|

# Good system
root@witherspoon:/tmp# hexdump -C witherspoon.pnor |more
00000000  50 41 52 54 00 00 00 01  00 00 00 01 00 00 00 80  |PART............|
```

A power cycle fixed the system (this was w37)

mdmillerii, 2017-05-19:
So `d0` is `50 << 1` so it would appear the data for the first byte is 1 clock off.

Do we limit the spi clock speed clock divider to the spec'd frequency? 

Also please capture the exact flash chip detected from dmesg controller probe.

geissonator, 2017-05-19:
Hit the exact same issue on w54:
```
[witherspoon]# pflash -i
Flash info:
-----------
Name          = /dev/mtd7
Total size    = 128MB 
Erase granule = 4KB 

TOC@0x00000000 Partitions:
-----------
FFS: Flash header not found. Code: 100
Error 100 opening ffs !
[witherspoon]# hexdump -C /dev/mtd7 |more
00000000  d0 41 52 54 00 00 00 01  00 00 00 01 00 00 00 80  |.ART............|
```

```
[witherspoon]# dmesg | grep controller
aspeed-smc 1e620000.flash-controller: Using IRQ 17
aspeed-smc 1e620000.flash-controller: DMA support deactivated.
aspeed-smc 1e620000.flash-controller: control register changed to: 00000640
platform 1e620000.flash-controller:flash@0: mx25l25635e (32768 Kbytes)
platform 1e620000.flash-controller:flash@1: mx25l25635e (32768 Kbytes)
aspeed-smc 1e630000.flash-controller: no DMA support.
aspeed-smc 1e630000.flash-controller: control register changed to: 00000740
platform 1e630000.flash-controller:flash@0: mx66l1g45g (131072 Kbytes)
i2c_aspeed 1e78a000.i2c: i2c controller registered, irq 25
```

```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-54"
VERSION_ID="v1.99.5-54-gb81bbef"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-54"
```

mine260309, 2017-05-22:
@mdmillerii A correction: `0xD0` is `1101 0000`, `0x50` is `0101 0000`, so `d0` is NOT `50 << 1`.

@geissonator I met similar but not identical issue on Palmetto. When the  Palmetto fails to boot host, the first 4 bytes read all `0xFF`. Power cycle "fixes" it as well.
Replacing the flash chip fixes the issue. For now I do not that the issue again.

But you mention this issue occurs on multiple Witherspoon systems, so it could be different. My comment is just FYI.

amboar, 2017-05-22:
@gwshan and I have also hit a very similar issue on Palmettos.

sivassrr, 2017-05-22:
Could able to hit it again today, First time, system booted fine.  
And passed the CT 142 TCs. 

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-60"
VERSION_ID="v1.99.5-60-gecfa662"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-60"
BUILD_ID="v1.99.5"
root@witherspoon:~# uname -a
Linux witherspoon 4.7.10-15c30d4deb0a01f398230395dc0d9af4be92d856 #1 Fri May 19 21:19:02 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# /usr/sbin/pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.15_1.22
	op-build-v1.15-115-ge46df0d-dirty
	buildroot-2017.02-4-gc28573e
	skiboot-5.5.0-rc3
	hostboot-3a3d65e
	linux-4.10.7-openpower1-pb32b856
	petitboot-v1.4.2-pa470686
	witherspoon-xml-07ef29d
	occ-e1889e5
	hostboot-binaries-46670e3
	capp-ucode-1bb7503
[==================================================] 100%
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:~# /usr/sbin/pflash -r /dev/stdout -P VERSION
FFS: Flash header not found. Code: 100
Error 100 opening ffs !
root@witherspoon:~# pflash -i
Flash info:
-----------
Name          = /dev/mtd7
Total size    = 128MB
Erase granule = 4KB

TOC@0x00000000 Partitions:
-----------
ID=00            part 00000000..00001000 (actual=00001000)
ID=01             HBB 00010000..000a0000 (actual=00090000) [ECC]
ID=02            HBEL 000a0000..000c4000 (actual=00024000) [ECC]
ID=03           GUARD 000d0000..000d5000 (actual=00005000) [ECC]
ID=04             HBD 000e0000..00200000 (actual=00120000) [ECC]
ID=05           DJVPD 00200000..00248000 (actual=00048000) [ECC]
ID=06            MVPD 00250000..002e0000 (actual=00090000) [ECC]
ID=07            CVPD 002e0000..00328000 (actual=00048000) [ECC]
ID=08             HBI 00330000..00f90000 (actual=00c60000) [ECC]
ID=09             SBE 00f90000..00fd8000 (actual=00048000) [ECC]
ID=10           HCODE 00fe0000..01100000 (actual=00120000) [ECC]
ID=11            HBRT 01100000..01580000 (actual=00480000) [ECC]
ID=12         PAYLOAD 01580000..01680000 (actual=00100000)
ID=13      BOOTKERNEL 01680000..02580000 (actual=00f00000)
ID=14           NVRAM 02580000..02610000 (actual=00090000)
ID=15             OCC 02610000..02730000 (actual=00120000) [ECC]
ID=16         FIRDATA 02730000..02733000 (actual=00003000) [ECC]
ID=17            CAPP 02740000..02764000 (actual=00024000) [ECC]
ID=18         SECBOOT 02770000..02794000 (actual=00024000) [ECC]
ID=19         BMC_INV 027a0000..027a9000 (actual=00009000)
ID=20            HBBL 027b0000..027b6000 (actual=00006000) [ECC]
ID=21        ATTR_TMP 027c0000..027c8000 (actual=00008000)
ID=22       ATTR_PERM 027d0000..027d8000 (actual=00008000) [ECC]
ID=23         VERSION 027e0000..027e1000 (actual=00001000)
ID=24     IMA_CATALOG 027f0000..02830000 (actual=00040000) [ECC]
ID=25         RINGOVD 02830000..02850000 (actual=00020000)
ID=26         WOFDATA 02850000..02b50000 (actual=00300000) [ECC]
ID=27     BACKUP_PART 03ff7000..03fff000 (actual=00000000)
root@witherspoon:~# /usr/sbin/pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.15_1.22
	op-build-v1.15-115-ge46df0d-dirty
	buildroot-2017.02-4-gc28573e
	skiboot-5.5.0-rc3
	hostboot-3a3d65e
	linux-4.10.7-openpower1-pb32b856
	petitboot-v1.4.2-pa470686
	witherspoon-xml-07ef29d
	occ-e1889e5
	hostboot-binaries-46670e3
	capp-ucode-1bb7503
[==================================================] 100%

sivassrr, 2017-05-22:
By mistake, got it closed.

geissonator, 2017-05-24:
Hit again on w37 this morning:
```
[witherspoon]# pflash -i
Flash info:
-----------
Name          = /dev/mtd7
Total size    = 128MB 
Erase granule = 4KB 

TOC@0x00000000 Partitions:
-----------
FFS: Flash header not found. Code: 100
Error 100 opening ffs !
[witherspoon]#  hexdump -C /dev/mtd7 |more
00000000  d0 41 52 54 00 00 00 05  00 00 00 01 00 00 00 80  |.ART............|

[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-76"
VERSION_ID="v1.99.5-76-g5327795"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-76"
BUILD_ID="v1.99.5"

```

mdmillerii, 2017-05-24:
@legoater there seems to be a consistent signature that the first byte is 0xd0 vs 0x50 expected.  I was corrected that this is 0x80 or'd into the data , so the first byte has the first bit set.

mdmillerii, 2017-06-26:
@legoater Could we be learning a speed that exceeds the chip spec?  Can we print the actual speed learned?

legoater, 2017-06-27:
@mdmillerii the system have seen with this issue was running with a 4.7 kernel which does not do any calibration to optimize the read. So the speed is the lowest.

legoater, 2017-07-02:
I have run a couple of read/write tests on a palmetto and on a witherspoon and could not reproduce :/

But the optimization read algo chooses a 100MHz SPI frequency which might be a bit too high for dual ouput IO on some chips (for the W25Q256 on palmetto but the MX66L1G45G on the witherspoon should be fine though). To do some tuning, we can introduce a property in the decice tree to  lower or increase the maximum SPI frequency used for a chip.

legoater, 2017-07-03:
see this initial patch https://github.com/legoater/linux/commit/91323ebf2a0981d1a9726c03ea393b3bcdfbf75f to lower the SPI frequency to 50MHz.

geissonator, 2017-09-18:
Something fixed this, haven't seen it in a while.  We also moved to UBI for witherspoon which changed this a lot as well.

closed 2017-09-18 by geissonator