openbmc/openbmc #1639: openbmc fails to build on mac os within docker container

geissonator, 2017-05-20:
I was hoping to get openbmc builds working on my mac.  The best option seemed to be to get docker running and use our build-setup.sh script. 

I did some tweaks to build-setup.sh to get things up and running (installing locales package and putting the loclae-gen calls after that) and some tweaks to $HOME (/Users vs. /home).  I'm somewhat at a loss on the bitbake error that I get though when building.  I tried for a few different machine configs but this seems to be a common recipe.

```
andrewg@moby:~/workspace/openbmc/build$ bitbake obmc-phosphor-image
Loading cache: 100% |#############################################################################################################################################| Time: 0:00:00
Loaded 801 entries from dependency cache.
WARNING: /home/andrewg/workspace/openbmc/import-layers/meta-openembedded/meta-oe/recipes-multimedia/jack/jack_git.bb: Exception during build_dependencies for waf_do_compile--:--
WARNING: /home/andrewg/workspace/openbmc/import-layers/meta-openembedded/meta-oe/recipes-multimedia/jack/jack_git.bb: Error during finalise of /home/andrewg/workspace/openbmc/import-layers/meta-openembedded/meta-oe/recipes-multimedia/jack/jack_git.bb
WARNING: /home/andrewg/workspace/openbmc/import-layers/meta-openembedded/meta-oe/recipes-multimedia/jack/a2jmidid_8.bb: Exception during build_dependencies for waf_do_compile
WARNING: /home/andrewg/workspace/openbmc/import-layers/meta-openembedded/meta-oe/recipes-multimedia/jack/a2jmidid_8.bb: Error during finalise of /home/andrewg/workspace/openbmc/import-layers/meta-openembedded/meta-oe/recipes-multimedia/jack/a2jmidid_8.bb
ERROR: ExpansionError during parsing /home/andrewg/workspace/openbmc/import-layers/meta-openembedded/meta-oe/recipes-multimedia/jack/a2jmidid_8.bb
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable waf_do_compile, expression was 	/home/andrewg/workspace/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/a2jmidid/8-r0/a2jmidid-8/waf build ${@get_waf_parallel_make(d)}
 which triggered exception IndexError: pop from empty list

Summary: There were 4 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

```
andrewg@moby:~$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="16.04.2 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.2 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
```

gtmills, 2018-05-31:
@geissonator Still have this problem or can we close?

geissonator, 2018-05-31:
Still doesn't work, one theory I heard was that Mac doesn't deal with case on it's directory paths, so that may be messing something up in yocto/linux docker.

mdmillerii, 2018-05-31:
@geissonator By default HFS+ is not case sensitive, but there is a mount option to create it case sensitive.  Typically you would do this on a separate data volume, although that creates disk partitioning issues.

That said, we should probably make sure at least our own repositories do not rely on case sensitivity.

geissonator, 2019-08-09:
I'm struggling to get the build-unit-test-docker.sh to work on my mac as well. It uses flock which is not supported on mac and also is having issues with this bash array for some reason:

```
+ PKG_REV=([boost]=1.69.0 [cereal]=v1.2.2 [CLI11]=v1.7.1 [fmt]=5.3.0 [function2]=e3695b4b4fa3c672e25c6462d7900f8d2417a417 [googletest]=9ab640ce5e5120021c5972d7e60f258bfca64d71 [json]=v3.3.0 [lcov]=75fbae1cfc5027f818a0bb865bf6f96fab3202da [linux-headers]=8bf6567e77f7aa68975b7c9c6d044bba690bf327 [libvncserver]=7b1ef0ffc4815cab9a96c7278394152bdc89dc4d [tinyxml2]=37bc3aca429f0164adf68c23444540b4a24b5778 [cppcheck]=df32b0fb05f0c951ab0efa691292c7428f3f50a9)
./build-unit-test-docker.sh: line 91: boost: unbound variable
```

dhruvibm, 2019-09-08:
I have been trying this for a while, any luck on this?

moto-timo, 2019-09-08:
Try using a Samba container and a bound volume for the file system.
See  https://github.com/crops/docker-win-mac-docs/wiki/Mac-Instructions

closed 2020-04-08 by stale[bot]