openbmc/openbmc #1641: [discussion] sdbusplus: support errors for property

mine260309, 2017-05-22:
sdbusplus currently only support errors for methods.
E.g. if a method defines an error:
```
methods:
    - name: Execute
      ...
      errors:
          - self.Error.CommandNotSupported
```
Then the generated cpp code will catch this exception:
```
    try
    {
        ...
        o->execute(convertCommandFromString(command));
        ...
    }
    catch(sdbusplus::internal_exception_t& e)
    {
        ...
    }
    catch(sdbusplus::xyz::openbmc_project::Control::Host::Error::CommandNotSupported& e)
    {
        ...
    }
```

However, it does not apply to getting/setting properties.

If getting/setting a property is not applicable (e.g. not allowed), it's preferable to throw an exception and letting the generated server code to handle the exception as how it is done for methods.

So what I want to discuss is:
1. Shall we add error support for properties?
2. If no, and getting/setting property does throw error exception, shall we add `catch (sdbusplus::exception_t& e)` for the server code so that it catches all the errors generated in `phosphor-dbus-interface`?
So the genrated server code will look like:
   ```
    catch(sdbusplus::internal_exception_t& e)
    {
        ...
    }
    catch(sdbusplus::exception_t& e)  // Let's catch all the generated exceptions in phosphor-dbus-interface
    {
        ...
    }
   ```
3. If still no, how do we handle errors on getting/setting properties?

williamspatrick, 2017-05-22:
1. Yes.

gtmills, 2018-05-03:
Setting a property like MacAddress `/xyz/openbmc_project/network/eth0/attr/MACAddress` should return an error if it fails to set. Currently, it returns okay even if it failed to set.

rfrandse, 2018-06-06:
<https://gerrit.openbmc-project.xyz/11016> property: Add errors support
Closes  #1641 sdbusplus: support errors for property

closed 2018-06-12 by geissonator