openbmc/openbmc #1642: Reboot via Host OS failed with quiesce state

sivassrr, 2017-05-22:
BMC and PNOR details:
```
- [x] BUILD ID   :  https://openpower.xyz/job/openbmc-build/751/
- [x] VERSION_ID : v1.99.5-60-gecfa662
- [x] Kernel     : Linux witherspoon 4.7.10-15c30d4deb0a01f398230395dc0d9af4be92d856 #1 Fri May 19 21:19:02 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.22
	op-build-v1.15-115-ge46df0d-dirty
	buildroot-2017.02-4-gc28573e
	skiboot-5.5.0-rc3
	hostboot-3a3d65e
	linux-4.10.7-openpower1-pb32b856
	petitboot-v1.4.2-pa470686
	witherspoon-xml-07ef29d
	occ-e1889e5
	hostboot-binaries-46670e3
	capp-ucode-1bb7503
```
**Problem:** 
One time system reboot via Host OS went fine. 
2nd time when tried, it failed as follows:

OS output:
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting /boot...
         Unmounting /run/user/0...
[  OK  ] Unmounted /run/user/0.
[  OK  ] Unmounted /boot.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped File System Check on /dev/d…8b97f-dd0f-4973-aa2d-db7484393160.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Stopped Create Static Device Nodes in /dev.
         Stopping Monitoring of LVM2 mirrors…ng dmeventd or progress polling...
[  OK  ] Removed slice system-systemd\x2dfsck.slice.
[  OK  ] Reached target Shutdown.
reboot: Restarting system
[  624.313322848,5] OPAL: Reboot request.

BMC Output:
```
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
```

sivassrr, 2017-05-22:
[hostosquiescejournalctl1642.txt](https://github.com/openbmc/openbmc/files/1018524/hostosquiescejournalctl1642.txt)

sivassrr, 2017-05-22:
Watchdog error is back. 

Apr 20 23:09:22 witherspoon host_watchdog.exe[775]: Polling watchdog: 0
Apr 20 23:09:22 witherspoon chassis_control.py[784]: Watchdog Error, Going to quiesce
Apr 20 23:09:22 witherspoon systemd[1]: Reached target Quiesce Target.
Apr 20 23:09:22 witherspoon phosphor-host-state-manager[880]: Auto reboot enabled. HOST BOOTCOUNT already set to 0.
Apr 20 23:09:22 witherspoon phosphor-host-state-manager[880]: Maintaining quiesce
Apr 20 23:09:22 witherspoon phosphor-host-state-manager[880]: Change to Host State

geissonator, 2017-05-22:
Yeah, didn't see this one, I'm sure it just a dup of #1645  I referenced #1645 in my code fix so we'll just have to dup this one to it even though it was first.

closed 2017-05-22 by geissonator