openbmc/openbmc #1643: obmc-fru-fault-monitor.service: Failed at step EXEC spawning  [Witherspoon]

gkeishin, 2017-05-22:
Driver info: v1.99.5-60-gecfa662
Build : https://openpower.xyz/job/openbmc-build/751/

On boot to BMC ready
```
Apr 20 17:30:08 witherspoon systemd[852]: obmc-fru-fault-monitor.service: Failed at step EXEC spawning /usr/sbin/phosphor-fru-fault-monitor: No such file or directory
Apr 20 17:30:08 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Main process exited, code=exited, status=203/EXEC
Apr 20 17:30:08 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Unit entered failed state.
Apr 20 17:30:08 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Failed with result 'exit-code'.
Apr 20 17:30:09 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Service hold-off time over, scheduling restart.
Apr 20 17:30:09 witherspoon systemd[1]: Stopped FRU Fault monitor service.
Apr 20 17:30:09 witherspoon systemd[1]: Started FRU Fault monitor service.

````

```
● obmc-fru-fault-monitor.service - FRU Fault monitor service
   Loaded: loaded (/lib/systemd/system/obmc-fru-fault-monitor.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-04-20 17:30:10 UTC; 2min 29s ago
  Process: 860 ExecStart=/usr/sbin/phosphor-fru-fault-monitor (code=exited, status=203/EXEC)
 Main PID: 860 (code=exited, status=203/EXEC)

Apr 20 17:30:09 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Main process exited, code=exited, status=203/EXEC
Apr 20 17:30:10 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Unit entered failed state.
Apr 20 17:30:10 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Failed with result 'exit-code'.
Apr 20 17:30:10 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Service hold-off time over, scheduling restart.
Apr 20 17:30:10 witherspoon systemd[1]: Stopped FRU Fault monitor service.
Apr 20 17:30:10 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Start request repeated too quickly.
Apr 20 17:30:10 witherspoon systemd[1]: Failed to start FRU Fault monitor service.
Apr 20 17:30:10 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Unit entered failed state.
Apr 20 17:30:10 witherspoon systemd[1]: obmc-fru-fault-monitor.service: Failed with result 'exit-code'.
root@witherspoon:~# 
```

gkeishin, 2017-05-22:
- [x] Journald 
[journald_v1.99.5-60-gecfa662_FRU_service.txt](https://github.com/openbmc/openbmc/files/1018869/journald_v1.99.5-60-gecfa662_FRU_service.txt)

spinler, 2017-05-22:
@dhruvibm I think you put this in?

dhruvibm, 2017-05-22:
Only the recipe was merged the app code is not yet merged.

https://gerrit.openbmc-project.xyz/#/c/3799/

geissonator, 2017-05-30:
This all fixed now @gkeishin?

gkeishin, 2017-05-30:
Yeah.. I dont see this now in latest build.. Closing this defect

closed 2017-05-30 by gkeishin