openbmc/openbmc #1644: Network suddenly stops working on witherspoon system

geissonator, 2017-05-22:
We've seen this intermittently on a few different systems.  It hit w34 today.  Everything looks fine via serial, but the eth0 IP address no longer pings.  A power cycle fixes the issue.  I verified the mac address and network settings look fine.

```
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-60"
VERSION_ID="v1.99.5-60-gecfa662"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-60"
BUILD_ID="v1.99.5"
```

```
root@witherspoon:~# dmesg | grep net
ftgmac100 1e660000.ethernet: Read MAC address 70:e2:84:14:20:33 from chip
ftgmac100 1e660000.ethernet: Using NCSI interface
ftgmac100 1e660000.ethernet eth0: irq 19, mapped at f0060000
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF 
```

```
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:20:33  
          inet addr:IP  Bcast:BCAST  Mask:255.255.255.0
          inet6 addr:IP Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6421506 errors:0 dropped:0 overruns:0 frame:2
          TX packets:384550 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:366309403 (349.3 MiB)  TX bytes:29848940 (28.4 MiB)
```

I know the network driver has been rewritten for 4.10 so we may just want to defer this until we get that in.
@gwshan

geissonator, 2017-05-22:
After the power cycle, network working again:
```
[witherspoon]# dmesg | grep eth
ftgmac100 1e660000.ethernet: Read MAC address 70:e2:84:14:20:33 from chip
ftgmac100 1e660000.ethernet: Using NCSI interface
ftgmac100 1e660000.ethernet eth0: irq 19, mapped at f0060000
ftgmac100 1e660000.ethernet eth0: NCSI interface up
```

gkeishin, 2017-05-23:
```
root@witherspoon:~# dmesg | grep net
ftgmac100 1e660000.ethernet: Read MAC address 70:e2:84:14:20:33 from chip
ftgmac100 1e660000.ethernet: Using NCSI interface
ftgmac100 1e660000.ethernet eth0: irq 19, mapped at f0060000
ftgmac100 1e660000.ethernet eth0: NCSI interface up
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
```
Another one network drop today..

gkeishin, 2017-05-23:
@geissonator Turns out the MAC on this system is 70:E2:84:14:20:72 .. Updating the system.

geissonator, 2017-06-02:
Haven't seen in a while, looks like it was another mac issue.

closed 2017-06-02 by geissonator