openbmc/openbmc #1645: op-wait-power-off@0.service  failed on OS reboot

gkeishin, 2017-05-22:
Boot to OS, and issue reboot

```
Apr 20 17:37:49 witherspoon phosphor-host-state-manager[880]: Received signal that host is off
Apr 20 17:37:49 witherspoon phosphor-host-state-manager[880]: Change to Host State
Apr 20 17:37:49 witherspoon phosphor-host-state-manager[880]: Reached intermediate state, going to next
Apr 20 17:37:49 witherspoon systemd[1]: Starting Disable UCD VCS rails...
Apr 20 17:37:49 witherspoon systemd[1]: Stopping Stop Power0...
Apr 20 17:37:49 witherspoon systemd[1]: Starting Wait for Power0 to turn on...
Apr 20 17:37:49 witherspoon systemd[1]: Stopped target Stop Host0.
Apr 20 17:37:49 witherspoon systemd[1]: Stopped Stop Watchdog0.
Apr 20 17:37:49 witherspoon systemd[1]: Stopped Stop Power0.
Apr 20 17:37:49 witherspoon systemd[1]: op-wait-power-off@0.service: Main process exited, code=killed, status=15/TERM
Apr 20 17:37:49 witherspoon systemd[1]: Stopped Wait for Power0 to turn off.
Apr 20 17:37:49 witherspoon systemd[1]: op-wait-power-off@0.service: Unit entered failed state.
Apr 20 17:37:49 witherspoon systemd[1]: op-wait-power-off@0.service: Failed with result 'signal'.
Apr 20 17:37:49 witherspoon phosphor-chassis-state-manager[883]: Received signal that power OFF is complete

```

```
● op-wait-power-off@0.service - Wait for Power0 to turn off
   Loaded: loaded (/lib/systemd/system/obmc-chassis-poweroff@0.target.requires/../op-wait-power-off@.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Thu 2017-04-20 17:37:49 UTC; 4min 3s ago
  Process: 1582 ExecStart=/bin/rm -f /run/openbmc/host@%i-on (code=exited, status=0/SUCCESS)
  Process: 1571 ExecStart=/bin/rm -f /run/openbmc/chassis@%i-on (code=exited, status=0/SUCCESS)
  Process: 1917 ExecStart=/usr/sbin/pgood_wait /org/openbmc/control/power%i off (code=killed, signal=TERM)
 Main PID: 1917 (code=killed, signal=TERM)

Apr 20 17:37:48 witherspoon systemd[1]: Starting Wait for Power0 to turn off...
Apr 20 17:37:49 witherspoon systemd[1]: op-wait-power-off@0.service: Main process exited, code=killed, status=15/TERM
Apr 20 17:37:49 witherspoon systemd[1]: Stopped Wait for Power0 to turn off.
Apr 20 17:37:49 witherspoon systemd[1]: op-wait-power-off@0.service: Unit entered failed state.
Apr 20 17:37:49 witherspoon systemd[1]: op-wait-power-off@0.service: Failed with result 'signal'.
root@witherspoon:~#

```

root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
root@witherspoon:~#

Driver Info

```
- [x] BUILD ID   : https://openpower.xyz/job/openbmc-build/751/
- [x] VERSION_ID : v1.99.5-60-gecfa662"
- [x] Kernel     : Linux witherspoon 4.7.10-15c30d4deb0a01f398230395dc0d9af4be92d856 #1 Fri May 19 21:19:02 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
root@witherspoon:~# cat /tmp/out.txt
IBM-witherspoon-ibm-OP9_v1.16_1.24
        op-build-v1.16-83-ge67b53c-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-rc2
        hostboot-d02273f
        linux-4.11.1-openpower1-pc11ec03
        petitboot-v1.4.2-pcb6424c
        machine-xml-e536539
        occ-8a7df03
        hostboot-binaries-d8f4ab7
        capp-ucode-9c73e9f
        sbe-d770027

```

- [x]  Journal log  
[journald_v1.99.5-60-gecfa662_OS_reboot_Quiesce.txt](https://github.com/openbmc/openbmc/files/1019093/journald_v1.99.5-60-gecfa662_OS_reboot_Quiesce.txt)

sivassrr, 2017-05-22:
Cool, With same BMC and little older PNOR, Hit with  similar error. https://github.com/openbmc/openbmc/issues/1642

gkeishin, 2017-05-22:
So this is pervasive on the old stable PNOR

- [x] Journald log for old stable PNOR test 
[journald_v1.99.5-60-gecfa662_OS_reboot_Quiesce_old_stable_PNOR.txt](https://github.com/openbmc/openbmc/files/1019158/journald_v1.99.5-60-gecfa662_OS_reboot_Quiesce_old_stable_PNOR.txt)

gkeishin, 2017-05-22:
This issue is pervasive on latest master build , I tried with new PNOR and old stable PNOR level combo to check on it.... so looks pretty broken to me... The last known good Master build
`Firmware Version:     v1.99.5-50-gbea7ab5 Build #748`

geissonator, 2017-05-22:
Looks like I broke this reboot path in https://github.com/openbmc/openbmc/commit/d75d9c516913bee640e3b4c245e8bd40e0965002

My goal was to ensure we call the obmc-host-stop services when someone calls obmc-chassis-poweroff.  The problem is that now both targets require each other which I'm assuming causes systemd to not require either (to avoid a cyclical dependency).  

One simple solution would be to put a check in phosphor-chassis-state-manager that would call the host-stop target if chassis-poweroff is called but that's adding logic to the application code we want to avoid, and would not fix the issue when a systems just uses systemctl to start the target.

Another solution would be to not allow the obmc-chassis-poweroff target to get called manually (i.e. just require people to always call obmc-host-stop target) but that doesn't fit well when someone is just manually powering on/off the system with the chassis target.

Or maybe I'm just over designing this whole thing and we should just require the user to know what they're doing when using these targets?  The system doesn't break when the host-stop services don't get run (basically disable watchdog).  I think for now, I'm going with this one.

geissonator, 2017-05-22:
Put up a revert here - https://gerrit.openbmc-project.xyz/#/c/4311/ that George helped to verify fixes the issue.

closed 2017-05-23 by williamspatrick