openbmc/openbmc #1646: Default boot count should only be set in one place

geissonator, 2017-05-22:
Currently the default boot count is set within pyphosphor/obmc/sensors.py on startup and in phosphor-state-manager/host_state_manager.cpp once quiesce state is the final state.  We need a design where we have the default boot count set in one place (currently both are hard coding to 3).

#1558 tracks refactoring of the sensors and getting them in dbus objects.  We can build on that once available to fix this issue.  

There's also been discussion on whether the host should really be controlling the boot count.  They have a bug preventing them from doing this currently but the design is that once a host boot is successful, they will come down and set the boot sensor back to it's default (so now we have 3 different places setting this default).  Not necessarily a part of this issue, but something we should consider.

geissonator, 2017-08-08:
# **STATUS**
8/8 - The refactoring has basically removed sensors.py so I think for the most part we're good now with this issue.

rfrandse, 2017-08-14:
<https://gerrit.openbmc-project.xyz/5851> Decrement reboot counter on all boot attempts
Resolves  #1646 Default boot count should only be set in one place

closed 2017-08-17 by williamspatrick