openbmc/openbmc #1649: OpenBMC flash instructions for EVB 2500 Board running AMI BMC firmware

mohitguptams, 2017-05-23:
folks,

where can i find instructions to update BMC with openBMC? I have AMI firmware on that BMC. I can boot to uboot. And my eval board has 32 MB flash. I can see multiple img-* files in my deploy/images/evb-ast2500 folder.  Which img file to use? I read somewhere i can use tftp, any instructions will help.

kc8apf, 2017-05-23:
I'm assuming you are currently using the u-boot that came with AMI firmware.  OpenBMC uses a different set of u-boot features so you'll need to replace it.  You'll want to use image-bmc which writes the entire flash (u-boot, kernel, initramfs, rootfs, r/w partition).  You'll probably need to use an external SPI programmer unless AMI's u-boot allows writing to the flash.

mohitguptams, 2017-05-24:
Thank you that answered my question.

closed 2017-05-24 by mohitguptams