openbmc/openbmc #1651: BMC filesystem: Create UBI read-only and read-write partitions

anoo1, 2017-05-24:


anoo1, 2017-06-01:
- [ ] Create service to attach UBI to BMC mtd.
- [ ] Create service to create UBI volumes (read-only and read-write).
- [ ] Change init script to mount ubi volumes.
- [ ] Mount read-write volume as overlay.

geissonator, 2017-06-15:
Adriana to split up into 2 stories, commit to this one

anoo1, 2017-07-11:
> Adriana to split up into 2 stories, commit to this one

Created these ones:
#1940 BMC: Ability for u-boot to execute the kernel that's in a ubi volume
#1943 BMC: Tell u-boot where squashfs image is

Lowering this story to 1 pt.

anoo1, 2017-07-11:
https://gerrit.openbmc-project.xyz/#/c/5168/ phosphor-software-manager: New service file to create ubi ro volumes
Resolves openbmc/openbmc#1651 BMC filesystem: Create UBI read-only and read-write partitions

stewart-ibm, 2017-07-13:
there should not be such a thing as /var/persist

```
Filesystem Hierarchy Standard:
> /var: Variable files—files whose content is expected to continually change during normal operation of the system—such as logs, spool files, and temporary e-mail files.
```

williamspatrick, 2017-07-13:
@stewart-ibm : Can you explain /var/lib then?  That is a very common place to put "persisted data that doesn't belong in etc".

williamspatrick, 2017-07-13:
```
/var/cache
Application cache data. Such data are locally generated as a result of time-consuming I/O or calculation. The application must be able to regenerate or restore the data. The cached files can be deleted without loss of data.
/var/lib
State information. Persistent data modified by programs as they run, e.g., databases, packaging system metadata, etc.
/var/mail
Mailbox files. In some distributions, these files may be located in the deprecated /var/spool/mail.
/var/opt
Variable data from add-on packages that are stored in /opt/.
```

These are all things that are expected to be persisted.

stewart-ibm, 2017-07-17:
Patrick Williams <notifications@github.com> writes:
> @stewart-ibm : Can you explain /var/lib then?  That is a very common
> place to put "persisted data that doesn't belong in etc".

That's where persistent data should live, not in /var/persist, but
rather FHS standard /var/lib and friends
-- 
Stewart Smith
OPAL Architect, IBM.

closed 2017-07-19 by williamspatrick