openbmc/openbmc #1657: OS failed to shutdown - NMI watchdog: BUG: soft lockup - CPU#1

gkeishin, 2017-05-24:
This is with drivers: FSI: Fix SBE/OCC for 2 socket properly https://gerrit.openbmc-project.xyz/#/c/4349/

```
- [x] SYSTEM     : ( Witherspoon - 2 Sockets - Pass 2 / 16 DIMMs )  
- [x] BUILD ID   : https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/3752/
- [x] VERSION_ID : v1.99.5-72-g30e4a1f
- [x] Kernel     : Linux witherspoon 4.7.10-fb92f38f80699b2089c3467b1344ac107f797d3c #1 Wed May 24 08:02:35 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:

IBM-witherspoon-ibm-OP9_v1.16_1.24
        op-build-v1.16-83-ge67b53c-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-rc2
        hostboot-d02273f
        linux-4.11.1-openpower1-pc11ec03
        petitboot-v1.4.2-pcb6424c
        machine-xml-e536539
        occ-8a7df03
        hostboot-binaries-d8f4ab7
        capp-ucode-9c73e9f
        sbe-d770027
```

Reboot from OS:

```
reboot: Restarting system
[  120.206898240,5] OPAL: Reboot request...
[  120.496436848,5] IPMI: Soft shutdown requested
INFO: rcu_sched detected stalls on CPUs/tasks:
        0-...: (2 ticks this GP) idle=60b/140000000000000/0 softirq=1524/1524 fqs=1048
        (detected by 19, t=2102 jiffies, g=1999, c=1998, q=54)
NMI watchdog: BUG: soft lockup - CPU#1 stuck for 22s! [kworker/1:1:1761]
INFO: rcu_sched detected stalls on CPUs/tasks:
        0-...: (2 ticks this GP) idle=60b/140000000000000/0 softirq=1524/1524 fqs=4191
        (detected by 19, t=8407 jiffies, g=1999, c=1998, q=54)
INFO: rcu_sched detected stalls on CPUs/tasks:
```

```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running

root@witherspoon:~# uptime
 17:51:49 up 22 min,  load average: 1.02, 1.07, 1.00
root@witherspoon:~#

Apr 20 17:36:29 witherspoon phosphor-host-state-manager[884]: Host State transaction request
Apr 20 17:36:29 witherspoon ipmid[864]: Transition request initiated successfully
Apr 20 17:36:29 witherspoon systemd[1]: Starting Wait for /org/openbmc/HostIpmi/1...
Apr 20 17:36:29 witherspoon systemd[1]: Started Wait for /org/openbmc/HostIpmi/1.
Apr 20 17:36:29 witherspoon systemd[1]: Starting Soft power off of the host...
Apr 20 17:36:30 witherspoon ipmid[864]: Pushing cmd on to queue
Apr 20 17:36:30 witherspoon ipmid[864]: Asserting SMS Attention
Apr 20 17:36:30 witherspoon ipmid[864]: SMS Attention asserted
Apr 20 17:36:30 witherspoon phosphor-softpoweroff[1184]: Host control signal values
Apr 20 17:36:30 witherspoon phosphor-softpoweroff[1184]: Timer started waiting for host to shutdown
```

gkeishin, 2017-05-24:
The same can be seen on Master build
```
- [x] BUILD ID   : https://openpower.xyz/job/openbmc-build/755/
- [x] VERSION_ID : v1.99.5-74-g60189dc
- [x] Kernel     : Linux witherspoon 4.7.10-fb92f38f80699b2089c3467b1344ac107f797d3c #1 Wed May 24 09:21:11 UTC 2017 armv6l GNU/Linux
```

geissonator, 2017-05-24:
This driver enabled soft power off which is where obmc sends a shutdown command to the host.  We'll need an issue opened against OPAL for this one (https://github.com/open-power/skiboot).  They're hanging in the shutdown path.  Could possibly use this bug to track a smaller timeout, we currently give them 45 minutes to power off.

gkeishin, 2017-05-24:
@geissonator Sure thing.. any data they would need apart from journald log ?

geissonator, 2017-05-24:
Nope, they may end up needing a system in the fail state but for FFDC, just the console, the PNOR version, and the OS level.

gkeishin, 2017-05-24:
Opened one for OPAL https://github.com/open-power/skiboot/issues/81

geissonator, 2017-05-24:
Workaround to do hard power off: systemctl start obmc-host-stop@0.target

mdmillerii, 2017-05-24:
@geissonator looking at the log in https://github.com/open-power/skiboot/issues/81 (and above) it shows that the OS has called opal to power off the system.  My quick trace of the code shows it should issue a `IPMI_CHASSIS_CONTROL` (`IPMI_NETFN_CHASSIS` (0x0) 0x2)   `IPMI_CHASSIS_HARD_RESET` (0x3), assuming that nested locks were not encountered.

@gkeishin please check the journal if that message was received?

geissonator, 2017-05-24:
Oh, I've got a bug here too (which maybe causes the opal shutdown one?)  We send up the soft power off, the host does it's shutdown and calls the BMC to power off, that then causes the soft power off function again which means we assert the attention again.  Sometimes things work, sometimes they don't it appears.

rfrandse, 2017-05-24:
https://gerrit.openbmc-project.xyz/4363 Revert "Use shutdown target to stop the host"
Resolves #1657 OS failed to shutdown - NMI watchdog: BUG: soft lockup - CPU#1

closed 2017-05-24 by williamspatrick