openbmc/openbmc #1661: Refactor mapper to be c++ application

geissonator, 2017-05-24:
Mapper is currently written in python, ideally it would be rewritten in c++ for code maintenance, performance, size, and service reasons.

geissonator, 2018-01-17:
Included in this is a rewrite of the command line tool used by the services.  See the review comments here for a proposal from Brad. https://gerrit.openbmc-project.xyz/#/c/2864/7/meta-phosphor/common/recipes-phosphor/chassis/obmc-op-control-power/op-reset-pgood-check%2540.service@12

gtmills, 2018-10-12:
@edtanous Has a C++ mapper rewrite. @geissonator Can we close?

bjwyman, 2019-07-24:
ibm-openbmc/dev#153 seems to imply that we now have a C++ mapper.

bjwyman, 2019-07-24:
Start: https://github.com/openbmc/phosphor-objmgr/commit/605206314f23710ef8a429745f139c891630588f

End: https://github.com/openbmc/meta-phosphor/commit/940ede4c9ababd85d5b766c3758439a00bb922b2

???

bjwyman, 2019-07-24:
Resolved by: https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/16321

anoo1, 2019-07-25:
Closing since it's resolved.

closed 2019-07-25 by anoo1