openbmc/openbmc #1663: Rename mboxd dbus object to be in correct xyz.openbmc_project

geissonator, 2017-05-24:
It's currently in org.openbmc.mboxd which is incorrect.  Will require service rename and minor code updates.

FYI - @anoo1 @dkodihal

geissonator, 2017-07-14:
Start with https://gerrit.openbmc-project.xyz/#/c/4739/

lgon, 2017-07-14:
Status Update
End Of Day July 7, 2017
------------------------------
Finished changing dbus name to xyz.openbmc_project.mboxd, and pushed to gerrit for review.

rfrandse, 2017-07-16:
<https://gerrit.openbmc-project.xyz/5318> mboxd: Rename mboxd dbus object to xyz.openbmc_project
Resolves  #1663 Rename mboxd dbus object to be in correct xyz.openbmc_project

lgon, 2017-07-21:
GitHub Comment
++++++++++++++++++++++++++++++++++++++
TASK LIST
- [ ] In phosphor-mboxd repository mplement c++ interfaces defined in .yaml file. https://gerrit.openbmc-project.xyz/#/c/4739/
- [ ] In moboxd repository, implement c interfaces defined in .yaml file.

#STATUS - In development

7-20 - No change in status. Will continue work on this item once #1550 is finished.
++++++++++++++++++++++++++++++++++++++

rfrandse, 2017-09-09:
<https://gerrit.openbmc-project.xyz/6578> mboxctl: Implemented mboxctl suspend method
Resolves  #1663 Rename mboxd dbus object to be in correct xyz.openbmc_project

geissonator, 2017-11-29:
As long as we filter org out of REST, this is low priority.

anoo1, 2017-11-29:
Per Brad, we'll just rename the interface to org.openpower.