openbmc/openbmc #1671: Phosphor-networkd:- Use case validation.

ratagupt, 2017-05-25:
ip address Create should return error(if dhcp is enabled )

ip address Delete should return error (user should not be allowed to delete the link local or dhcp ip address).

when users creates the ip it may happen that ipaddress and prefix is same but gatway is changed so we need to change the gateway accordingly.

ipaddress/gateway/subnet mask(v4/v6) validation.

Need to write a utility which test that the given gateway is reachable/not.User should add the gateway which is reachable.

ratagupt, 2017-08-21:
We should only validate that the IP/gatway is in ithe inet_pton format http://man7.org/linux/man-pages/man3/inet_pton.3.html

ngorugan, 2017-09-21:
- [ ] Update ip address Create to 

     1. return error,if dhcp is enabled
     2. change the gateway if ipaddress and prefix is same but gatway is changed while user creating ip.

- [ ] ip address Delete should return error for link local and dhcp ip addresses

- [ ] Write test cases to Validate ipaddress/gateway/subnet mask(v4/v6).

- [ ] Write a utility which tests that the given gateway is reachable/not.

rfrandse, 2017-10-03:
<https://gerrit.openbmc-project.xyz/7048> IP Address use case validation
Resolves  #1671 Phosphor-networkd:- Use case validation.

closed 2017-10-11 by geissonator