openbmc/openbmc #1674: Sprint 17.05.31 - Committed

rfrandse, 2017-05-25:


dkodihal, 2017-05-29:
Squad 1 sprint plan

**Jayanth**
**17.05.31**
#1510 BMC Dump: Add infrasture support for User initiated Dump

**Devender**
**17.05.31**
#1560 IPMI FRU commands : implementation

**Vishwa**
**17.05.31**
#1634 Generate error log on Host Watchdog timeout

**Tom**
**17.05.31**
#1633 Add support for hwmon labels to phosphor-hwmon 
#1515 SEL: Implement the mandatory SEL commands

anoo1, 2017-05-30:
# Squad 3 -

**Adriana:anoo1 -**

**17.05.31**
- BMC filesystem

---

**Andrew:geissonator -**

**17.05.31**

---

**Michael:mtritz -**

**17.04.05**
- #1351 pflash: Add option to retrieve PNOR partition flags
  * 05/18: Moved to Actual.

**17.05.31**
- #1554 PNOR: Implement Software ActivationProgress.interface

---

**Saqib**

**17.05.31**
- #1553 PNOR: Implement Software RedundancyPriority.interface

---

**Leo**

**17.05.31**
-

bradbishop, 2017-05-30:
-------------------------------------------------------------------------------------------------
***Squad 2 -***
-------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------
**Gunnar:gtmills** 
**17.05.31:**
- #940 (fan floor)

-------------------------------------------------------------------------------------------------
**Matt S:spinler** 
**17.05.31:**
- #1342 (Watchdog support)

-------------------------------------------------------------------------------------------------
**Matt B:msbarth**
**17.05.31:**
- #1626 (calculate speed change request)

-------------------------------------------------------------------------------------------------
**Brad:bradbishop**
**17.05.31:**
- #1318 (gpio presence)
- #1300 (real error logs in presence)

-------------------------------------------------------------------------------------------------
**Chris:cbostic**
**17.05.03:**
- #1599 (perf improvements)

**17.05.17:**
- #1177 (Add IR35221 to Witherspoon device tree)
   - 05/25 - vcs workaround is in the way of doing this - blocker

**17.05.31:**
- #1635 (Non hwmon attrs for MAX)
    - 05/26 - coded, will start testing

- #1636 (watchdog driver sysfs)
     - 05/25 - blocked on sysfs vs DT discussion
------------------------------------------------------------------------------------------------
**Eddie:eddiejames**
**17.03.22:**
- **#1330 (OCC: Set label attribute)**
    - 05/22 - same state as #1611.

**17.05.31:**
- #1619 (non hwmon OCC sysfs attributes)

-------------------------------------------------------------------------------------------------
**Brandon:bjwyman**

**17.05.17:**
- **#1265 (hwmon-iio)**
   - 05/25 - pushed up a change that finds the path, looking at callouts

-------------------------------------------------------------------------------------------------
**Dinesh:chinaridinesh**
**17.03.08:**
- #1315 (Ensure uboot fw_env update script uses new inventory object) 
   - 05/26 - bring to design meeting, getting different options

**17.05.03:**
- #1322 (pgood is lost on BMC reboot on witherspoon)

**17.05.31:**
- #1555 (make real error logs in fan control)
   - 05/30 - up for review

closed 2017-06-26 by zahrens